don/cordova-plugin-ble-central #274: android program java.lang.NullPointerException

wzhsh90, 2016-07-12:
plugin version:1.1.1

PluginManager: Uncaught exception from plugin
                                                                java.lang.NullPointerException
                                                                    at com.megster.cordova.ble.central.Peripheral.registerNotifyCallback(Peripheral.java:318)
                                                                    at com.megster.cordova.ble.central.Peripheral.processCommands(Peripheral.java:626)
                                                                    at com.megster.cordova.ble.central.Peripheral.queueCommand(Peripheral.java:592)
                                                                    at com.megster.cordova.ble.central.Peripheral.queueRegisterNotifyCallback(Peripheral.java:568)
                                                                    at com.megster.cordova.ble.central.BLECentralPlugin.registerNotifyCallback(BLECentralPlugin.java:413)
                                                                    at com.megster.cordova.ble.central.BLECentralPlugin.execute(BLECentralPlugin.java:195)
                                                                    at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:117)
                                                                    at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:98)
                                                                    at org.apache.cordova.PluginManager.exec(PluginManager.java:133)
                                                                    at org.apache.cordova.CordovaBridge.jsExec(CordovaBridge.java:59)
                                                                    at org.apache.cordova.engine.SystemExposedJsApi.exec(SystemExposedJsApi.java:41)
                                                                    at com.android.org.chromium.base.SystemMessageHandler.nativeDoRunLoopOnce(Native Method)
                                                                    at com.android.org.chromium.base.SystemMessageHandler.handleMessage(SystemMessageHandler.java:27)
                                                                    at android.os.Handler.dispatchMessage(Handler.java:102)
                                                                    at android.os.Looper.loop(Looper.java:136)
                                                                    at android.os.HandlerThread.run(HandlerThread.java:61)

don, 2016-07-13:
@wzhsh90 I need more details in order to duplicate this. What were you doing when this error happened? Does this happen every time? What device were you connected to?

don, 2016-08-18:
@wzhsh90 If you can provide more info on how to duplicate this problem comment on the issue and I'll re-open it.

closed 2016-08-18 by don