don/cordova-plugin-ble-central #276: startNotification() is slow to receive new characteristics -- Ionic 2

nlgreen, 2016-07-13:
I'm making an Ionic 2 app that connects to an Arduino and receives rapid data notifications. The [nRF Connect app](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) is able to be notified of 50+ data points in a second or less, but using startNotification() is much slower at a few data points per second maximum (with the same Arduino code). Is there a way to define the data exchange interval to be faster?

The application code is very simple:

```
// scan and connect to a peripheral...

connected(peripheral) {

        /* Subscription for the custom service */
        let subscription = BLE.startNotification(peripheral.id, this.scanInfo.service, this.scanInfo.char);

        subscription.subscribe(buffer => {
            let data = new Uint32Array(buffer);
            console.log(data[0]); 
        });
}
```

don, 2016-07-14:
I'm not sure. Can you share your Arduino code, details about your Arduino hardware, Android hardware, and Android software? I'll take a look.

nlgreen, 2016-07-14:
- [Arduino 101](https://www.arduino.cc/en/Guide/Arduino101) with [CurieBLE](https://www.arduino.cc/en/Reference/CurieBLE)
- Nexus 6P running Android 6.0.1
- Moto G running Android 5.1 (Both phones have same issue)

I made some short code examples with the problem I've been experiencing:

[Arduino Code](https://github.com/mdash-ursi2016/arduino_code/blob/master/speed_test.ino). It should just send 250 numbers via Bluetooth as fast as it can once a connection is found.

And some very simple Ionic 2 to log the numbers:

```
export class HelloIonicPage {
    constructor() {
	this.scanInfo = { service: 'aa7b3c40-f6ed-4ffc-bc29-5750c59e74b3',
                          heartrate: 'b0351694-25e6-4eb5-918c-ca9403ddac47'
                        };
    }

    connect() {
        BLE.scan([],5).subscribe(device => {
            if (device.name == "Penelope") {
                BLE.connect(device.id).subscribe(() => {
                    BLE.startNotification(device.id, this.scanInfo.service, this.scanInfo.heartrate).subscribe(buffer => {
                        var data = new Uint8Array(buffer);
                        console.log(JSON.stringify(data));
                    });
                }, error => {
                    console.log("Peripheral disconnected");
                });
            }
        });
    }
}

```

don, 2016-07-15:
@nlgreen I can't duplicate what you're seeing. I ran some tests and I get the same results in nRF Connect and Cordova.

| Device  | OS | Native Timing | Cordova Timing |
| ------------- | ------------- |------------- |------------- |
| Nexus 5X  | Android 6.0.1  | 22 s | 22 s|
| Galaxy S4 GPE  | Android 5.1  | 24 s | 24 s|
| iPhone 6s  | iOS 9.3.2  | 15 s | 15 s|

Your sketch started sending data as soon as the central connected. I modified [the code](https://github.com/don/ble-issue276/blob/master/arduino/Curie101Notification/Curie101Notification.ino) to begin sending notifications after the central subscribes. I ran this on an Arduino 101.

Native timings were run using [nRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) on Android and [LightBlue](https://itunes.apple.com/us/app/lightblue-explorer-bluetooth/id557428110?mt=8) on iOS. The sketch prints the time to the serial console.

Note that since you're using notification, the central is free to ignore or drop them if it receives too many. On iOS I've seen problems if notifications are sent faster than every 30ms.

code: https://github.com/don/ble-issue276

don, 2016-07-27:
@nlgreen any updates? Can I close this?

nlgreen, 2016-07-28:
Thanks for all the help you provided on this. Unfortunately the problem persisted (in a strange manner—the send rate starts very rapid and slows down after 100-200 packets sent) but if the problem couldn't be duplicated I'm sure it was on our end. I'll close it.

closed 2016-07-28 by nlgreen

chjchoi, 2018-01-18:
I used the CC2541 System and stack 1.4.2.2 for ble.
It worked very good on Android 4.4.4 system.
but on Android 5.  it was very very slow...

I have found out and resolved the problem.
CC2541 I used 16bit UUID  such as 'fff0' 'fff1'
I switched the service uuid and characteritcs uuid to 128bit uuid
the slow problem was resolved.

Instead of 'fff0' 'fff1'  I used the  
0000FFF0-0000-1000-8000-00805F9B34FB
0000FFF1-0000-1000-8000-00805F9B34FB

this uuid was from other BLE scan app's displaying...
I have not changed the 16bit uuid system on CC2541
because when I use the 'fff0' 'fff1'  it means the 128uuids above.

More exact result
I should change the CC2541 uuid systems to 128bit
and use TI BASE UUID 
F000FFF0-0451-4000-B000-000000000000
F000FFF1-0451-4000-B000-000000000000
and change the Ionic uuid also 
but I did not changed the CC2541 stack uuid.
just use the 0000 ....34FB 128bit uuid only on the ionic app.

woodz-, 2018-05-22:
@nlgreen, did you get it to your satisfactory?
I am currently struggling with CC41-A Module (HM-10 clone) connected to Android 7 on Samsung SIII I9300.

- Cordova: 7.0.1
- node: v8.9.4
- npm: 6.0.1
- cordova-plugin-ble-central 1.1.9 "BLE"

Having a notification set up with the ble-central plugin (service: "ffe0", characteristic: "ffe1"), the first 30 secs comm is working very well. Then gets stuck to only 2 notifications per second on about 100ms in distance. On the embedded side every 400ms a frame of 32 ascii-bytes is generated and transmitted to the RX-pin @9600 baud.

Before I came to ble, I was working with cordova-plugin-bluetooth-serial 0.4.7 "Bluetooth Serial" with same equipment given above - no problems there.

Greetings woodz

9600, 2018-05-22:
You called?

don, 2018-05-22:
@woodz- is your radio dual-mode? Can you disable Bluetooth classic and see if it works better?

woodz-, 2018-05-23:
Got some news around the comm stuck. Did an IOS build and ran on an iPhone 4s - IOS 9.3.x. Was removing the "cordova-plugin-bluetooth-serial 0.4.7" since it led to crashes on iPhone and was anyway just hanging around.
A comm test over 2 hours worked like a charm. No comm stuck. The embedded equipmend is the same like when I had problems on Android.
So there must be somthing with the S3 or A7. I am not sure how to explicitly turn off classic Bluetooth in A7 settings, since there are just the Bluetooth on / off and scanning settings. I might remove the bt-serial Plugin as I did on ios and give it another try.
Whether the radio is dual mode, I would say yes related to the S3 A7 phone, since it requests for pairing on classic bt and scans and connects to the HM-10 BLE module.
Will keep you informed...

woodz-, 2018-05-26:
After removing the "cordova-plugin-bluetooth-serial 0.4.7" from the android build and testing on 2 different devices

1. Samsung Galaxy S3 I9300 | Android 7.1.2 | Bluetooth 4.0, A2DP, EDR, aptX
2. Vodafone VF-795 | Android 5.1 | Bluetooth 4.0, A2DP

the ble comm on notifications still gets stuck after a few seconds.
Was trying a serial ble monitor (from play store) on both devices, to see whether they show the same behaviour -> they do! Because of facing this, one could argue, it's not an issue of the "ble-central" plugin. It could moreover have something to do with android itself (over different versions).
Which is kind of frustrating, because how would I know, whether to get it to work alt all on android?
Both devices have Bluetooth 4.0 as its hardware. Will try to obtain a device with Bluetooth 4.1. Then tell you again...

Greetings woodz

woodz-, 2018-06-04:
After obtaining a S6 with bluetooth 4.1 and Android 7.0, the original coming with this phone I am kind of discouraged. It's showing the same symptoms as all the Android phones I have to test.
So, what I would be interested in, am I the only one struggling with those issues?
It looks like there is something with the software interface between cordova <-> bluetooth or android <-> cordova.
My intention is to write a native test application to see whether the HM-10 clone gets working over the different gatt functions. But this will take time.

don, 2018-06-05:
Does this happen with every peripheral or just one? There could be a problem with the periphreal or firmware. Enable the HCI debug log in Android settings and view the results in Wireshark to see where the Bluetooth communication is getting stuck.

woodz-, 2018-06-06:
Thanks for answering don.
I have another 4 pieces of those peripherals but they are all from the same supplier coming all together at once. But yes, I will test them all through.
One question: how could we explain your assumption (faulty firmware) in contrast to the fact that all went perfectly fine on IOS? It's the same peripheral, same plugin but different phone and different OS...
Will post my test results soon.

woodz-, 2018-06-07:
I was recording a 15secs comm session.
- The result can be found here:
[btsnoop_hci.log](https://github.com/don/cordova-plugin-ble-central/files/2078520/btsnoop_hci.log)
- Periodically there is an error reported on ATT packets:
Expert Info (Error/Protocol): Frame is out of any "connection handle" session
- The position, where the comm gets stuck is on/after frame No. 110:
110	6.864599	controller	host	HCI_EVT	13	Rcvd LE Meta (LE Connection Update Complete)

Would be interesting to see a log of IOS in contrast to compare. Anybody knows how to do hci logs on IOS without doing jailbreak?

Greetings woodz

mj2sdev, 2021-01-08:
Hi, looking at the problem, I stopped swimming because it was similar to the problem I had. I think it has already been solved, but I think we can process the data output part with ngZone.