don/cordova-plugin-ble-central #277: Retrieving data from kCBAdvDataServiceData

eztrac, 2016-07-14:
Hi 
I am confused about how to get data retrieved from kCBAdvDataServiceData. I do get correct byteLength but when I try to access kCBAdvDataServiceData['FED8'].data, I get nothing. Any suggestion

{
    "name": "demo",
    "id": "D8479A4F-7517-BCD3-91B5-3302B2F81802",
    "advertising": {
        "kCBAdvDataChannel": 37,
        "kCBAdvDataServiceData": {
            "FED8": {
                "byteLength": 7 // data not shown
            }
        },
        "kCBAdvDataLocalName": "demo",
        "kCBAdvDataServiceUUIDs": ["FED8"],
        "kCBAdvDataManufacturerData": {
            "byteLength": 7  // data not shown
        },
        "kCBAdvDataTxPowerLevel": 32,
        "kCBAdvDataIsConnectable": true
    },
    "rssi": -53
}

don, 2016-07-14:
The device data in an ArrayBuffer. You need to put it into a typed array.

    if (cordova.platformId === 'ios') {
       if (device && device.advertising &&
           device.advertising.kCBAdvDataServiceData &&
           device.advertising.kCBAdvDataServiceData.FED8) {
               data = new Uint8Array(device.advertising.kCBAdvDataServiceData.FED8);
       }
    }

Other service data might be a Uint32

    if (cordova.platformId === 'ios') {
       if (device && device.advertising &&
           device.advertising.kCBAdvDataServiceData &&
           device.advertising.kCBAdvDataServiceData.FABC) {
               hardwareId = new Uint32Array(device.advertising.kCBAdvDataServiceData.FABC)[0];
       }
    }

eztrac, 2016-07-14:
Thanks it worked.

closed 2016-07-14 by eztrac