don/cordova-plugin-ble-central #278: Working on iOs, unable to read on Android

twinsit, 2016-07-14:
I have an ionic app that is working great on iOs, but on Android I have some issues on Android.
The app can connect to my device, and I get the peripheralData:

> {
      "name": "Bike",
      "id": "F9:C7:BA:2D:37:D5",
      "advertising": {},
      "rssi": -99,
      "services": [
      "1800",
      "1801",
      "180a",
      "58de1000-4083-4735-8926-6721a778ae5e"
  ],
  "characteristics": [
    {
      "service": "1800",
      "characteristic": "2a00",
      "properties": [
        "Read",
        "Write"
      ]
    },
    {
      "service": "1800",
      "characteristic": "2a01",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "1800",
      "characteristic": "2a04",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "1801",
      "characteristic": "2a05",
      "properties": [
        "Indicate"
      ],
      "descriptors": [
        {
          "uuid": "2902"
        }
      ]
    },
    {
      "service": "180a",
      "characteristic": "2a29",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "180a",
      "characteristic": "2a24",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "180a",
      "characteristic": "2a25",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "180a",
      "characteristic": "2a27",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "180a",
      "characteristic": "2a26",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "180a",
      "characteristic": "2a28",
      "properties": [
        "Read"
      ]
    },
    {
      "service": "58de1000-4083-4735-8926-6721a778ae5e",
      "characteristic": "58de1101-4083-4735-8926-6721a778ae5e",
      "properties": [
        "Notify"
      ],
      "descriptors": [
        {
          "uuid": "2901"
        },
        {
          "uuid": "2902"
        },
        {
          "uuid": "2904"
        }
      ]
    },
    {
      "service": "58de1000-4083-4735-8926-6721a778ae5e",
      "characteristic": "58de1102-4083-4735-8926-6721a778ae5e",
      "properties": [
        "Read",
        "Write"
      ],
      "descriptors": [
        {
          "uuid": "2901"
        },
        {
          "uuid": "2904"
        }
      ]
    },
    {
      "service": "58de1000-4083-4735-8926-6721a778ae5e",
      "characteristic": "58de1103-4083-4735-8926-6721a778ae5e",
      "properties": [
        "Read"
      ],
      "descriptors": [
        {
          "uuid": "2901"
        },
        {
          "uuid": "2904"
        }
      ]
    },
    {
      "service": "58de1000-4083-4735-8926-6721a778ae5e",
      "characteristic": "58de1104-4083-4735-8926-6721a778ae5e",
      "properties": [
        "Read",
        "Write",
        "Indicate"
      ],
      "descriptors": [
        {
          "uuid": "2901"
        },
        {
          "uuid": "2902"
        },
        {
          "uuid": "2904"
        }
      ]
    },
    {
      "service": "58de1000-4083-4735-8926-6721a778ae5e",
      "characteristic": "58de1105-4083-4735-8926-6721a778ae5e",
      "properties": [
        "Notify"
      ],
      "descriptors": [
        {
          "uuid": "2901"
        },
        {
          "uuid": "2902"
        },
        {
          "uuid": "2904"
        }
      ]
    },
    {
      "service": "58de1000-4083-4735-8926-6721a778ae5e",
      "characteristic": "58de1106-4083-4735-8926-6721a778ae5e",
      "properties": [
        "Read"
      ],
      "descriptors": [
        {
          "uuid": "2901"
        },
        {
          "uuid": "2904"
        }
      ]
    },
    {
      "service": "58de1000-4083-4735-8926-6721a778ae5e",
      "characteristic": "58de1107-4083-4735-8926-6721a778ae5e",
      "properties": [
        "Read",
        "Write"
      ],
      "descriptors": [
        {
          "uuid": "2901"
        },
        {
          "uuid": "2904"
        }
      ]
    },
    {
      "service": "58de1000-4083-4735-8926-6721a778ae5e",
      "characteristic": "58de1108-4083-4735-8926-6721a778ae5e",
      "properties": [
        "Read",
        "Write"
      ],
      "descriptors": [
        {
          "uuid": "2901"
        },
        {
          "uuid": "2904"
        }
      ]
    },
    {
      "service": "58de1000-4083-4735-8926-6721a778ae5e",
      "characteristic": "58de1109-4083-4735-8926-6721a778ae5e",
      "properties": [
        "Read"
      ],
      "descriptors": [
        {
          "uuid": "2901"
        },
        {
          "uuid": "2904"
        }
      ]
    }
  ]
}

Then I read using this part of code 

> BLE.read(this.connectedDevice,"58DE1000-4083-4735-8926-6721A778AE5E","58DE1102-4083-4735-8926-6721A778AE5E").then(information => {
var data = new Uint32Array(information);
console.log("--------- TRIP 002 -----------"+JSON.stringify(data));
},error => { this.errorlogger("trip 002: "+JSON.stringify(error));	});	

On iOs this works great. But on Android I doesn't get any data, even no error...
Any suggestions (I tried lowercase/uppercase already)
					`

don, 2016-07-14:
Check `adb logcat` for stack traces. Check chrome://inspect for JavaScript errors (or use cordova-plugin-console.) If you don't see anything there, enable HCI debugging in the Android developer preferences to see what's happening.

don, 2016-08-19:
@twinsit did you get this problem resolved? Do you think it might be related to the bonding issue in #293

twinsit, 2016-08-23:
I was able to resolve this. I had some issues with bluetooth on the phone I think, and after some reboots it suddenly worked, and now it still works :-)

closed 2016-08-23 by twinsit