don/cordova-plugin-ble-central #280: Notifications limited to 50Hz on iOS ?

LarsBuur, 2016-07-15:
Currently we are connecting to our device using a custom ansi-c program using gatt-tool to download the data from our device.

Our device is custom build and track movement and more in 20Hz and save it to flash storage. Afterwards we connect to the device and I have implemented an Cordova app using cordova-plugin-ble-central@1.1.0 plugin and Cordova iOS 4.1.0

Everything runs smooth until I run into a data session with more than 18000 samples (about 15 minutes of data) and it takes about 90 seconds before the problem occurs. This suggest that we are streaming data in about 18000/90 =~ 200 Hz =~ 5 ms each. (and that we are able to collect the first 90 seconds correctly)

I have found  one closed issue where someone suggested that it might be possible to "keep the connection" alive sending a read/write command - this has no effect and I have tried both using setInterval every second. iOS refuses to keep the connection open after 90 seconds if it receives  notification data faster than 50Hz.  

Trying to reconnect to the notification characteristic makes it restart - so thats not an option either. 

In issue #246 @djlitvak suggests that the problem can be/is related to our device streaming notifications faster than 50Hz - and I am sure that we are streaming the collected data as fast as we can. 

This blog post (https://punchthrough.com/blog/posts/maximizing-ble-throughput-on-ios-and-android) suggests that the limit of 50Hz.(20ms.) specified by Apple is not even correct - but it is more likely to be around 33Hz(30ms.) 

Do you see any other way of fixing this issue but change the firmware. We have a fair share of devices deployed so this is not an ideal solution. 

Thank you very much for a great module.

Lars

don, 2016-07-15:
BLE doesn't work well for large data transfers. I have an [old project](https://github.com/don/rfduino-logreader/) where we needed to add a [30ms delay between notification](https://github.com/don/rfduino-logreader/blob/42c6f72e0f95a433b4640d9ca5231940e415462e/Arduino/RFduinoFileUploader/RFduinoFileUploader.ino#L71) or iOS would skip some notifications. (Notifications are not reliable so it's legitimate for iOS to drop/ignore them.) We determined that 30ms worked best using trial and error. It's possible you can send data faster on newer hardware.

Can you try introducing a delay? Maybe a delay between every notification is enough, or perhaps a more significant delay every X notifications? I haven't found good documentation on this so I think you'll need to determine via experimentation.

For my most recent abuses BLE, we are sending files from the phone to the peripheral. We use write (as opposed to writeWithoutResponse) to guarantee we get all the data. 

Using an **indication** instead of a notification on your peripheral should do the same thing. It will throttle the data and guarantee you receive all of it. Unfortunately sending the confirmation slows down the transfer.

We found that increasing the characteristic size to 500 speeds up file transfer a lot (maybe 6x) on iOS. The plugin transparently handles the multiple reads required for larger packets. (Actually it is the OS handling this.) Unfortunately many Android devices can't handle 500 byte characteristics. For those that can, larger characteristics made the transfer a bit slower (IIRC about 10%).

LarsBuur, 2016-07-18:
Thank you very much for your reply Don. 

I have tried all I can think of, client side. 

I will forward your suggestions to our hardware people.

Best,
Lars

closed 2016-07-18 by LarsBuur