don/cordova-plugin-ble-central #281: startNotification error on iOS

BramHendriksDeveloper, 2016-07-20:
I am trying to subscribe to a certain Notify event. However, it does not seem to work. This is how I use the function:

```
var testFunction = function(password){
            var callBack = function(buffer){
                console.log("notify succes");
                var data = new Uint8Array(buffer);
                console.log("State changed to " + data[0]);
            }
            ble.startNotification(services.deviceId, services.password.service, services.password.notification, callBack);
        };
```

The error given in xCode goes as following:

> registering for notification
> 2016-07-20 13:44:10.287 SafeToBike[684:189315] getData
> 2016-07-20 13:44:10.288 SafeToBike[684:189315] -[NSNull compare:options:range:]: unrecognized selector sent to instance 0x19ff41490
> 2016-07-20 13:44:10.288 SafeToBike[684:189315] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[NSNull compare:options:range:]: unrecognized selector sent to instance 0x19ff41490'
> *** First throw call stack:
> (0x181fb6db0 0x18161bf80 0x181fbdc4c 0x181fbabec 0x181eb8c5c 0x187d38098 0x187d384d8 0x1000213a8 0x10001c72c 0x10004bdf0 0x10004b69c 0x10004cc58 0x18298c02c 0x181f6d09c 0x181f6cb30 0x181f6a830 0x181e94c50 0x18377c088 0x18717e088 0x10001a088 0x181a328b8)
> libc++abi.dylib: terminating with uncaught exception of type NSException

don, 2016-07-20:
Did you scan and connect to the peripheral? Can you ensure that the characteristic supports notify?

Check to make sure you're not passing a null value id, service, or characteristic due to a typo or something being out of scope.

BramHendriksDeveloper, 2016-07-21:
Solved the problem. I feel ashamed but it is fixed now. I was passing a null object indeed since I discarded some changes including the characteristic value.

closed 2016-07-21 by BramHendriksDeveloper

don, 2016-07-21:
Glad it's working. The plugin code could  probably use some null checking and have better warnings or error messages.

subratpalhar92, 2020-09-07:
make sure you are using write() instead of writeWithoutResponse(), this solved my issue on ios