don/cordova-plugin-ble-central #282: ble.write() not working on Android

GianlucaCesari, 2016-07-20:
Hi there @don ,
I've been trying to use your plugin in my app, and is working fine on iOs, the only thing is when I try to write on a characteristic on the method ble.write(), the first parameter (device.id) doesn't work in android (even if the mac address is the right one). Tell me if you need more information or if is just something I've been doing wrong.

don, 2016-07-20:
Do you have an error message? Are you using the MAC address you got from the ble.scan()?

GianlucaCesari, 2016-07-20:
Yes, I'm using the MAC address coming from ble.connect(),  the error message is the general error which I console.logged in the error function. The error says "Write failed."

don, 2016-07-20:
The exact same code works on iOS with the same peripheral?

What hardware and what version of Android? What version of this plugin?

Use chrome://inspect to see if there are JavaScript errors (requires Android 5+) 

Run 'adb logcat' and see if there's an error message in the log. 

It's possible you have some JavaScript that works on iOS and fails on Android.

GianlucaCesari, 2016-07-20:
Yes, it works on iOS with the same peripheral(an arduino 101), I'm using a samsung for android (I believe it's a galaxy s5 with the latest update(I will check tomorrow morning)). I've already checked for js errors but there are no errors. I will try 'adb logcat' first thing tomorrow morning and I will report here the results.

GianlucaCesari, 2016-07-21:
[image]
[image]
[image]
So this is the code, hope it can help, the android device is a S5 with Android 5.0 onboard

don, 2016-07-21:
'chars[0].service' is the problem. On Android, you're writing to the wrong characteristic. Android returns the Generic Access Service 0x1800 and the Generic Attribute Service 0x1801 in addition to your custom service. (iOS hides the 0x1800 and 0x1801 services)

I recommend explicitly stating the service and characteristic UUID in your Cordova code.

GianlucaCesari, 2016-07-21:
Oh, that was the problem, thank you very much for your help!

closed 2016-07-21 by GianlucaCesari

jsanchez1993, 2018-10-12:
I Dom, I have the same problem but your answer is not correct for my. 
this sis my error: Characteristic 00001801-0000-1000-8000-00805f9b34fb not found.
`
var data = new Uint8Array(1);
		data[0] = 97;
		ble.write(app.device_id, '1800', '1801', data.buffer, function (success) {
			console.log('success: ');
			console.log(success);
		}, function (failure) {
			console.log('error: ');
			console.log(failure);
		});`