don/cordova-plugin-ble-central #283: strange behaviour on iOs

hervethouzard, 2016-07-21:
Hello,

We are using our own application (for Android and iOs) with our own peripheral (a BLE peripheral used to open doors).

On my application I have a button to ask for the door to  open.
On Android I can click as much as I want on the button, the door will open, each time.

On iOs, it fails one in two.
The first time, the door opens, the second time fail.
And the next time it runs and the next time it will fail.

Where it's strange it's when it fails, the second time, that's just after the call to ble.connect()
Both callbacks, connectSuccess and connectFailure, are called...!!???

Do you have any idea of what's happening?

Bye,
Hervé

hervethouzard, 2016-07-21:
To give you more information on the process used by the application:
1) The application starts by scanning for BLE peripheral during 6 seconds
2) It creates a list of BLE Peripherals founded in the nearby
3) Then it stops to scan
3) The application search for the "better" peripheral to connect to (according to the RSSI) 
4) It connects to the most "near by" BLE peripheral
5) We activate notifications (indications in our cases)
6) We dialog witht the BLE peripheral
7) We stop indications
8) We disconnect

Here is the full "process"

don, 2016-07-21:
Can you get the log output from Xcode?

hervethouzard, 2016-07-21:
Is there a way to send you the log without publishing it here?

don, 2016-07-21:
email address is in github profile

hervethouzard, 2016-07-22:
Hello,

mail sent.
Thank you.

closed 2016-08-18 by don