don/cordova-plugin-ble-central #284: ble  can not scan in background

wzhsh90, 2016-07-22:
ble plugin version:1.1.1
cordova platform :ios 4.2.0

test step:

 1、call the method  ble.startScan([], function (device) {},function(){}) and then lock the screen
 
test result:
 1、the plugin can not scan ble in background

 2、connected to the ble  device , then lock the screen the plugin work well

don, 2016-07-25:
iOS can scan in the background but I haven't done it with this plugin, so I'm not exactly sure of the behavior. (I do know that background notifications work fine.)

1) You must enable [UIBackgroundModes](https://github.com/don/cordova-plugin-ble-central#background-notifications-on-ios)

2) IIRC you need to scan for a specific UUID rather than any UUID

don, 2016-08-19:
I've confirmed that scanning, connecting and notifications work in the background. See sample project https://github.com/don/ble-background for more details.

Duplicate of #255

closed 2016-08-19 by don

Hypher, 2017-11-19:
I didnt manage to make it work on background on iOS 10 as stated here: https://github.com/don/ble-background/issues/1

Deepankerchaudhary, 2017-12-11:
I am trying to scan BLE Device through background. Plugins are working fine for Android in background but in iOS it's not working.
When I try to scan in background through startScan([serviceUDID],success, failure);
Sometimes later it's timeout in iOS. Without timeout it always scanning...
I have tried background plug-in separately as well but nothing is happening.
Kindly help!