don/cordova-plugin-ble-central #285: iOS:CoreBluetooth didWriteValueForCharacteristic returning an old value

manaswinik, 2016-07-22:
The data which we are write is not updating the characteristic value and when i call read characteristic response gives me the old value.
   Stuck with this issue from many days any help would be great.Thanks in advance

don, 2016-07-22:
Test the same write the read using another app like LightBlue Explorer on iOS or nRF Connect on Android. 

Can you provide more details? Does this happen on iOS or Android? What versions? Sample code helps too.

manaswinik, 2016-07-25:
Hi 
  Thanks for the reply. This issue is on iOS and i have tried using app LightBlue Explorer and i noticed the same behaviour. No matter what value i write on the characteristic the value doesn't change. Characteristic value that i am using is

"properties":[
"Read",
"Write"
],
"isNotifying":false,
"characteristic":"49535343-6DAA-4D02-ABF6-19569ACA69FE",
"service":"49535343-FE7D-4AE5-8FA9-9FAFD205E455"

And the code that i am using for write is 

[peripheral writeValue:data forCharacteristic:aChar type:CBCharacteristicWriteWithResponse];

the error value in didWriteValueForCharacteristic method is nil which i am assuming to be write is success and when i am trying to read the value from the same characteristic with code

        [peripheral readValueForCharacteristic:characteristic];

the response is not expected. And for the same peripheral device i could successfully connect, write an read with android device

don, 2016-07-25:
@manaswinik since the problem is also happening with LightBlue, I'm thinking this isn't a plugin problem.

Could it be related to the data you're sending? 

Do you have any visibility to debug the peripheral or at least sniff the BLE packets and see what's different between iOS and Android?

closed 2016-08-18 by don