don/cordova-plugin-ble-central #287: Prompt to turn on bluetooth on Android

myyellowshoe, 2016-07-25:
When the app is first started and bluetooth is off shouldn't the library prompt the user to turn it on? It does this on IOS fine, but doesn't seem to do it on android.

Running android 4.3. With Crosswalk on a Samsung S3.

don, 2016-07-25:
@myyellowshoe currently the best way to do this is [ble.startStateNotifications](https://github.com/don/cordova-plugin-ble-central#startstatenotifications). You'll get an initial state notification when the apps starts and whenever the state changes. 

On Android you can prompt the user to enable Bluetooth when necessary. This is the behavior I'd prefer on iOS too, although currently the user is prompted.

myyellowshoe, 2016-07-28:
Perfect. That's exactly what I needed. Just simply added the ble.enable on start. So it prompts them to turn on bluetooth on app start. As in our case it's required for the app to function.

closed 2016-07-28 by myyellowshoe