don/cordova-plugin-ble-central #289: BLE Connection time irregular

hoongoon86, 2016-07-29:
Hi, 

I'm trying to connect a BLE device to the app I'm developing. 
It basically works all good but the connection time is very random. 
When it connect right away, it takes only 1~2 seconds to connect.
Some other times, it takes more than 10 seconds.(or more)

For details, here's my code for connection.

`
   ble.isEnabled(function(){ 

        function ble_scan(){

            ble.startScan([], function(ble_device) {

                if(ble_deviceId == ""){

                    // When first time connecting to device, find by matching name 
                    var tempName = "";

                    if(deviceType == "iOS"){
                        if(ble_device.name == ble_deviceName){
                            tempName = ble_device.advertising.kCBAdvDataLocalName;
                        } else if(ble_device.advertising.kCBAdvDataLocalName == ble_deviceName){
                            tempName = ble_device.name;
                        }
                    } 

                    else {
                        tempName = ble_device.name;
                    }

                    if(tempName == ble_deviceName){
                        checkScanCompleted = true;
                        ble_deviceId = ble_device.id;   
                        ble_saveId(ble_deviceId);
                    }

                } 

                else {
                    if(ble_device.id == ble_deviceId){
                        checkScanCompleted = true;
                    } 
                } 
            }, false);

            // check if it scanned
            var bleScanned = setInterval(function(){
                if(checkScanCompleted == true){
                    ble.stopScan();
                    ble_connect();
                    clearInterval(bleScanned);
                }
            },5);
        }

        function ble_connect(){
            var checkDeviceConnected = false;

            ble.connect(ble_deviceId, function(){
                checkDeviceConnected = true;
            }, false);

            var bleConnect = setInterval(function(){
                if(checkDeviceConnected == true){
                    intro_fadeControl();        // Goto next html page and 
                    ble_startReadWrite();     // start write,read
                    clearInterval(bleConnect);
                }
            },5);

        }

        ble_scan();

    },function(){
        finishLoading();
    });`

don, 2016-07-29:
You have 2 intervals of 5 seconds each, so worst case you could see 10 seconds.

I'd try removing the intervals and moving that logic into the callback.

don, 2016-07-29:
Maybe something like this. (Not tested)

        ble.isEnabled(function(){

            function ble_scan(){

                ble.startScan([], function(ble_device) {

                    if(ble_deviceId === ""){

                        // When first time connecting to device, find by matching name
                        var tempName = "";

                        if(deviceType == "iOS"){
                            if(ble_device.name == ble_deviceName){
                                tempName = ble_device.advertising.kCBAdvDataLocalName;
                            } else if(ble_device.advertising.kCBAdvDataLocalName == ble_deviceName){
                                tempName = ble_device.name;
                            }
                        }

                        else {
                            tempName = ble_device.name;
                        }

                        if(tempName == ble_deviceName){
                            ble_deviceId = ble_device.id;
                            ble_saveId(ble_deviceId);
                            ble.stopScan();
                            ble_connect();
                        }
                    }

                    else {
                        if(ble_device.id == ble_deviceId){
                            ble.stopScan();
                            ble_connect();
                        }
                    }
                }, false);

            }

            function ble_connect(){

                ble.connect(ble_deviceId, function(){
                    intro_fadeControl();        // Goto next html page and
                    ble_startReadWrite();     // start write,read
                }, false);
            }

            ble_scan();

            },function(){
                finishLoading();
            }
        );

hoongoon86, 2016-07-29:
Thank you for your answer. 

I was trying to check scanned / connected states every 5ms. 
I'll try 'not nested' way of connecting and let you know! :)

don, 2016-07-29:
Oh, yeah that is 5ms. Read it wrong.

Add some logging to see if the scanning or connecting is taking a long time.

Try using another tool like LightBlue or nRF Connect to discover and connect to the device. It might not advertise very often, or perhaps it's just slow when you connect.

hoongoon86, 2016-07-29:
I don't see the difference. 
Still when it connects good, it's good(~2s). When it's not, takes ~10s. 
With LightBlue, it connects instantly. :D 
I was surprised that it takes less than 1s. (tested multiple times)

don, 2016-07-29:
When the plugin connects, it discovers all the services, then all the characteristics for services. This can be slow on some devices especially if there are a lot of devices or characteristics. 

You should be able to see some log output in Xcode or from adb logcat as services and characteristics are being discovered. 

In a native app, the solution is to search for only the services and characteristics you need, not everything. There's no option in the plugin, although you could modify the native code to do this.

hoongoon86, 2016-07-29:
From xcode, I see the log says,
'Connecting to peripheral with UUID : '
and it sits on this state for awhile (of course when it takes long).
Scanning, however takes 1~1.5 second with the result of 4 scanned devices around. 

I'll fix the native and test with it!  
Thanks for your support!

hoongoon86, 2016-08-02:
Hi, 

Bad news is, I don't see any difference with connectivity time. 
This could be because I tested with 1~3 BLE devices around but anyway...
If it always maintain slow connection time, I understand that there's a problem.
But when it's in a good mood, it connect quite fast. 
Any other thoughts or suggestions I can try? 

Like always, thank you for your support! :)

hoongoon86, 2016-08-02:
Hi, 

Bad news is, I don't see much difference. 
Maybe because only few devices were around when tested but anyway....
If it's always maintaining slow connection time, I understand that there's a problem.
But when it's in a good mood, it connects quite fast.
Any other thoughts or recommendation I can try? 

Always thank you for your support :)

hoongoon86, 2016-08-09:
Hi, 

Found out what caused the issue. As Don predicted, the problem was because there were too many services and characteristics in a device. (3 services with 20 characteristics)
For the record, (for those who are struggling with the same issue I had)
Here's the code to specify what to scan. 

`- (void)centralManager:(CBCentralManager *)central didConnectPeripheral:(CBPeripheral *)peripheral {

    NSLog(@"didConnectPeripheral");

    peripheral.delegate = self;

    // NOTE: it's inefficient to discover all services
    NSArray *m_service = [NSArray arrayWithObjects:
                                  [CBUUID UUIDWithString:@"YOUR_UUID"],nil];
    
    [peripheral discoverServices:m_service];

    // NOTE: not calling connect success until characteristics are discovered
}`

And do the same for 'discoverCharacteristics'

closed 2016-08-09 by hoongoon86

don, 2016-08-09:
@hoongoon86 thanks for the update. modifying the plugin is the best solution for now. Perhaps in the future version there'll be a way to pass just the services and characteristics to discover.

itechdom, 2017-09-18:
@hoongoon86 thank you so much for your proposed solution. It worked very well for me. 

@don Thanks for the great plugin! I have an app that works on Android too. Any tips on how I can limit the services discovered on Android?

Thanks!

don, 2017-09-18:
@itechdom you should be able to add similar code on Android to only discover the services and characteristics you care about.