don/cordova-plugin-ble-central #290: Issue with 24 Byte Chracteristic

unclespode, 2016-07-29:
Hard to pin point to to a particular issue, but I have a 24 byte characteristic. Trying to read it always returns an empty array buffer. Writing to it is erratic - and whatever is written is nothing like what I'm sending it - a 24 part array buffer. I've only tested this on Android - using another plugin I had no issues (well, I did, but different ones - it still worked).

Any ideas? I have a feeling I just need to tell me firmware guys that we need to only do 20 bytes, as nothing seems very well configured for >20.

don, 2016-07-29:
Are you sure the array buffer is empty when you read it? Make sure you create a new typed array from the array buffer.

    function onReadData(buffer) {
        var data = new Uint8Array(buffer);
        console.log(data);
    }

Keeping characteristics to 20 bytes or less may be a good idea. This plugin can handle large characteristics values.  I have some 80 byte characteristics that work reliably across iOS and Android, but you need the peripheral to support this. The plugin relies on the iOS and Android doing blob writes for larger characteristics.

When you're writing, make sure you send an ArrayBuffer

    function writeData() {
        var data = new Uint8Array([1,2,3,4,5]);
        ble.write(device_id, service, characteristic, data.buffer, success, failure);
    }

I'd also use [nRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) to read and write the same data to your characteristic. This will help determine if the problem is the app code or the peripheral.

don, 2016-08-19:
@unclespode did you resolve this?

unclespode, 2016-09-05:
Apologies - I went back to a different plugin, so no I have not.

closed 2016-09-17 by don