don/cordova-plugin-ble-central #291: set notification failed for ...

locsim, 2016-08-02:
Hi everyone,

I've been playing with this plugin for a few days, trying to have a cordova app and a BLE device talk to each other. The device is a so called "itag", a keyring bought on ebay, purpose of which is ~= "don't loose your keys".

What I could do: 
-coonect to the device and get some of its characteristics
-read the state of the button and also the battery level using ble.read(..)
-make the device ring using ble.writeWithoutResponse(..)

What I could not do (and this is the reason for this post):
-be notified when the device button is pressed.

ble.startNotification(device.id, "ffe0", "ffe1", success, fail) result is fail function answering: 
"set notification failed for 0000ffe1-0000-1000-8000-00805f9b34fb"

Here are the device characteristics published when connected:

{"name":"MLE-15","id":"FF:FF:F0:00:7D:A2","advertising":{},"rssi":-56,
"services":["1800","180f","1802","ffe0"],
"characteristics":[
{"service":"1800","characteristic":"2a00","properties":["Read","Notify"]},										
{"service":"1800","characteristic":"2a01","properties":["Read"]},												
{"service":"180f","characteristic":"2a19","properties":["Read","Notify"]},										
{"service":"1802","characteristic":"2a06","properties":["WriteWithoutResponse","Write","Notify"]},				
{"service":"ffe0","characteristic":"ffe1","properties":["Read","Notify"]}]}										

Here is the code:
`ble.scan([], 5, scan_success, scan_fail);
				
		function scan_success(device) {
			ble.connect(device.id, connect_success, connect_fail);
		}
		function scan_fail() {
			alert('error during scan');
		}
		
		function connect_success(device) {
			console.log(JSON.stringify(device));
			ble.startNotification(device.id, "ffe0", "ffe1", success, fail);
		}
		function connect_fail(device) {
			alert('error during connect');
		}		
						
						
		var success = function(buffer) {
			// Decode the ArrayBuffer into a typed Array based on the data you expect
			var data = new Uint8Array(buffer);
			alert("device answers: " + data[0]);
			}
		
		var fail = function(buffer) {
			alert(buffer);
			}		`
Any help appreciated!

locsim, 2016-08-08:
Hello,
Did anyone succeed to get a MLE-15 itag to work using this plugin?
(I mean use the itag button to trigger anything in the app)

Thank you!

don, 2016-08-09:
@locsim it looks ffe1 has the notify property and you're subscribing to the notification properly. Make sure there are no JavaScript errors in your app with jshint. Use chrome://inspect or safari -> develop -> phone -> index.html and make sure there are no runtime errors.

I'd also see if you can use nRF Connect or Light Blue Explorer to connect to the device and receive notifications.

Is this the device you're using? https://www.amazon.com/Bluetooth-Wireless-Anti-lost-Compatible-Smartphones/dp/B017BDPMYC/ref=sr_1_1?ie=UTF8&qid=1470780201

locsim, 2016-08-11:
Hi Don,

Thank you for your answer.
-No js error reported by Chrome,
-I checked with an android app (com.macdom.ble.blescanner).
Well, I could not do more using this app than I could do with your plugin (read RSSI, name and characteritics and read battery level).

This is very strange to me because the app supposed to be used with the keyring is notified by the keypress, which means there is a way. But which way?

-Yes, this is the device I'm using.

don, 2016-08-18:
@locsim startNotification is failing because this device doesn't expose the [Client Characteristic Configuration Descriptor 0x2902](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml). The DFRobot Bluno [has the same problem](https://github.com/don/cordova-plugin-ble-central/issues/251#issuecomment-226369279).

You can see this using LightBlue on iOS. iTag is on the left. The descriptors are missing. RFduino does this correctly and 0x2902 is visible on the right.

[image]

Since the iTag doesn't meet spec, it's not going to work with the plugin. Assuming that notifications are always enabled on the device, you can probably modify the plugin to make it work.

For Android comment out the code that returns an error.
https://github.com/don/cordova-plugin-ble-central/blob/0eba7febc0a6dcfb8befae05337df22fddf13438/src/android/Peripheral.java#L354

For iOS comment out the code that tries to write to the descriptor
https://github.com/don/cordova-plugin-ble-central/blob/dde35f8b03088b78b140752fb2133f5d61a15265/src/ios/BLECentralPlugin.m#L190

closed 2016-08-18 by don

locsim, 2016-08-23:
> 2902_50

Hi Don,

Thank you very much for your help, I do appreciate.
FYI, this one works ok:
http://www.ebay.com/itm/201410181915  (seebelow its "signature")
Kind regards from south of France,
Philippe

{"name":"KF01","id":"FF:FF:00:00:B8:59","advertising":{},"rssi":-52,

"services":["1800","1801","180a","1802","1804","1803","180f","ffe0"],

"characteristics":[
{"service":"1800","characteristic":"2a00","properties":["Write"]},
{"service":"1800","characteristic":"2a01","properties":["Read"]},
{"service":"1800","characteristic":"2a04","properties":["Read"]},
{"service":"1800","characteristic":"2a02","properties":["Read"]},
{"service":"1800","characteristic":"2a03","properties":["Read"]},
{"service":"1801","characteristic":"2a05","properties":["Indicate"],"descriptors":[{"uuid":"2902"}]},
{"service":"180a","characteristic":"2a29","properties":["Read"]},
{"service":"180a","characteristic":"2a24","properties":["Read"]},
{"service":"180a","characteristic":"2a25","properties":["Read"]},
{"service":"180a","characteristic":"2a26","properties":["Read"]},
{"service":"180a","characteristic":"2a27","properties":["Read"]},
{"service":"180a","characteristic":"2a28","properties":["Read"]},
{"service":"180a","characteristic":"2a23","properties":["Read"]},
{"service":"180a","characteristic":"2a2a","properties":["Read"]},
{"service":"180a","characteristic":"2a50","properties":["Read"]},
{"service":"1802","characteristic":"2a06","properties":["WriteWithoutResponse"]},
{"service":"1804","characteristic":"2a07","properties":["Read","Notify"],"descriptors":[{"uuid":"2902"},{"uuid":"2904"}]},
{"service":"1803","characteristic":"2a06","properties":["Read","Write"]},
{"service":"180f","characteristic":"2a19","properties":["Read","Notify"],"descriptors":[{"uuid":"2902"},{"uuid":"2904"}]},
{"service":"ffe0","characteristic":"ffe1","properties":["Notify"],"descriptors":[{"uuid":"2902"},{"uuid":"2901"}]}
]
}