don/cordova-plugin-ble-central #292: Does it work between two iPhone devices?

liangqiangkun, 2016-08-03:
I want to communicate between two iPhone devices instead of iPhone and Arduino，Can I  realize  with it ？If it can works， please tell me the solution，thanks very much！

don, 2016-08-09:
This plugin provides the Bluetooth central role, connecting to a Bluetooth low energy service. This plugin will connect to another iPhone if the 2nd iPhone is advertising a Bluetooth Low Energy service. This plugin won't create that service. 

I can another plugin, http://github.com/don/cordova-plugin-ble-peripheral that allows you to create peripherals on iOS.  See the UART example for communitcating between 2 phones. https://github.com/don/cordova-plugin-ble-peripheral/tree/master/examples/uart

closed 2016-08-09 by don

liangqiangkun, 2016-08-26:
OK,thanks!