don/cordova-plugin-ble-central #293: Pin request working on iOs, not on Android

twinsit, 2016-08-04:
Unfortunatly in the last phase before releasing the client decided to work now with a pin code on the device. For iOs this is working great, the pin popup shows up, and then I can give in the pin + submit, and everything works.

On Android I don't get that popup. Is there something I need to do additional?

Thanks

Wouter

don, 2016-08-04:
I don't know. What Android phone and version are you using? What bluetooth device are you connecting to? Is the PIN code because of read/write encrypted permissions on some characteristics? Did you try bonding to the device from the Android Bluetooth settings screen?

twinsit, 2016-08-04:
I use a Hawaii p9 lite. The bluetooth device is an e-bike. When I try to connect on iOs he asks it to me directly the PIN (when I do the first read action). When I try this in the Nordic Connect app on android, he asks me the pin when I try to read. In my code it doesn’t. I try tomorrow once if I can see the device in the Bluetooth Setting screen). But in fact it should work like on iOs, that he asks me the pin in the app (and remember it then).

don, 2016-08-19:
@twinsit did you get this working?

Also on my Huawei device Cordova works a lot better once I updated the [System WebView](https://play.google.com/store/apps/details?id=com.google.android.webview)

don, 2016-08-19:
You could also try the reading a characteristic value to get the device to prompt for a pin (See https://github.com/don/cordova-plugin-ble-central/issues/127#issuecomment-167808782)

twinsit, 2016-08-23:
I did some updates and when I do now a read he requests the PIN. So this is fixed! Thanks

closed 2016-08-23 by twinsit