don/cordova-plugin-ble-central #294: ble.startScan not providing name in callback array

rrpatil761, 2016-08-06:
I got the following response 

Object {advertising: ArrayBuffer, id: "61:9F:57:28:87:EF", rssi: -84}

Please help.

rrpatil761, 2016-08-08:
Any Update please?

don, 2016-08-08:
This is fine. Some devices don't have names or advertise names.

rrpatil761, 2016-08-09:
But I have scanned at least 20 mobile devices but none of them having name :(

don, 2016-08-09:
It's possible there's something with your phone or none of the devices have names. Send me a link to your code I can try running on some of my devices.

rrpatil761, 2016-08-10:
Hi Don,

Thanks For quick reply,

Please check following code,

  $scope.StartScan = function () {
        console.log('start scan');
        ble.startScan([], onScan, isnotenabled);
    }

    function onScan(peripheral) {
        if ($filter('filter')($scope.availabledevices, peripheral.id) == "") {
            $scope.availabledevices.push(peripheral);
        }
        console.log($scope.availabledevices);
    }

In peripheral object there is no name key available, please verify and let me know.

don, 2016-08-16:
That code snippet looks OK, but I'm not going to build an angular project.  I still think you're not seeing names because none of your devices have names.

You could try https://github.com/don/ITP-BluetoothLE/tree/master/phonegap/scan

    git clone https://github.com/don/ITP-BluetoothLE/
    cd phonegap/scan
    cordova plugin add cordova-plugin-ble-central 
    cordova plugin add cordova-plugin-dialogs
    cordova plugin add cordova-plugin-console
    cordova platform add android
    cordova run android --device

closed 2016-08-18 by don