don/cordova-plugin-ble-central #296: no results ble.startScan() on Android SDK 23

dsansoni, 2016-08-11:
Hi, I'm running a cordova (v6.2.0) project on an Android Marshmallow device (with SDK 23).
I'm using the version 1.1 of this plugin but ble.startScan() does not show any result.

I found a temporary workaround using target-sdk 22 but I'd like to know if there's a real solution to this problem.

Sincerly,
Davide

don, 2016-08-11:
Version 1.1 supports android-23 and should prompt the user for location access which is required for BLE scanning. Check the app settings to see if this permission has been disabled. Alternately you can delete and re-install the app and it should prompt you again.

dsansoni, 2016-08-12:
Delete and re-install the app solved the issue. Thank you for your prompt reply

closed 2016-08-12 by don