don/cordova-plugin-ble-central #299: Plug-in takes lot of time to connect BLE device very first time

unnivm, 2016-08-16:
I am using your plug-in to connect to BLE device. The device name is 'Qtemp' from  'Qtemp.co'. It is taking almost 30 seconds while connecting to this BLE device very first time. But subsequent connections are faster.

I would appreciate if you can provide me any hint to this issue.

Thanks

don, 2016-08-16:
Are you seeing this on iOS or Android? I'd guess service discovery is slow on the phone or peripheral and something is cached the 2nd time you connect.  Look at #289 for suggestions on how to modify the code to find only services and characteristics you care about.

unnivm, 2016-08-16:
Hello Don,

I am using Android version 6 with Nexus phone.

Thanks for your input.

Unni M

unnivm, 2016-08-17:
When it is connecting to BLE device, the following command is used:

gatt = device.connectGatt(activity, true, this,
BluetoothDevice.TRANSPORT_LE);

The above code connects to BLE device in background mode which is the
default one. This background connection has scan interval of 1280ms and a
window of 11.25ms and if is not scanning, connection may take more time to
complete it.

So it is a good idea to change the connection mode to "direct" where scan
interval is very less.

gatt = device.connectGatt(activity, false, this,
BluetoothDevice.TRANSPORT_LE);

Please let me know your thoughts on this.

Thanks
Unni M

don, 2016-08-19:
I suspect the scan connection time might be a duplicate of #289. You might be able to restrict the services and characteristics that are discovered to make this faster.

**autoConnect** is currently *false* [in the code](https://github.com/don/cordova-plugin-ble-central/blob/99d5d6b0778ac0d092373040a1303248c4f2ba32/src/android/Peripheral.java#L69-L71) so you should get direct connection when using this plugin. Where'd you see the line with autoConnect true?

unnivm, 2016-08-19:
Please find the method below in "Perpheral.java":

public void connect(CallbackContext callbackContext, Activity activity) {
        BluetoothDevice device = getDevice();
        connectCallback = callbackContext;
        if (Build.VERSION.SDK_INT < 23) {
            gatt = device.connectGatt(activity, false, this);
            //gatt = device.connectGatt(activity, true, this);
        } else {
            gatt = device.connectGatt(activity, false, this,
BluetoothDevice.TRANSPORT_LE);
            //gatt = device.connectGatt(activity, true, this,
BluetoothDevice.TRANSPORT_LE);
        }

Earlier, it was true while connecting to device. Now I commented that line.

Thanks
Unni M

don, 2016-08-19:
@unnivm the plugin's default is to use autoConnect false. See [Peripheral.java](https://github.com/don/cordova-plugin-ble-central/blob/99d5d6b0778ac0d092373040a1303248c4f2ba32/src/android/Peripheral.java#L69-L71) I think works well for most cases. I'm not sure where the version came from with autoConnect true. Maybe those were your local mods? 

For the connection speed issue, I'd try on different devices to make sure you're not seeing a phone specific problem. Ideally your app should be able to connect about as quickly as the app provided with the hardware. Some peripherals are slow. For peripherals with lots of services, discovery is slow, especially on iOS. One trick is to limit which services and peripherals the plugin discovers, since discovering everything is slow. Issue #289 should have some tips on how you can modify the plugin for your specific hardware.

closed 2016-08-19 by don

unnivm, 2016-08-19:
Don, I appreciate your input. Could you please tell me where should I make
such changes in Android case?

Thanks a lot.

Unni M

don, 2016-08-19:
On closer inspection it doesn't look like you can limit the service discovery on Android. 

You could profile the code to see if the discovery is the slow part. The request happens in [gatt.discoverServices](https://github.com/don/cordova-plugin-ble-central/blob/45c3566b5e411d717630cc2148ab531e96398d74/src/android/Peripheral.java#L215). The response come back in [onServicesDiscovered](https://github.com/don/cordova-plugin-ble-central/blob/45c3566b5e411d717630cc2148ab531e96398d74/src/android/Peripheral.java#L193).

If you are working on the QTemp app send me an email (my address is in my github profile.)

unnivm, 2016-08-19:
Thank you Don for your inputs. I will email to you directly and update you
the progress.

Thanks
Unni M