don/cordova-plugin-ble-central #301: [Question] IOS / MBP scan issue

tchauviere, 2016-08-22:
Hi,

I'm currently trying to scan for my MBP Bluetooth from my PhoneGap app with your plugin, but it seems that I have nothing that shows up during scan. My bluetooth is supposed to be LMP V4.0 so I guess it should work ?

Any idea about that ?

Thanks a lot and for this nice plugin :)

Best regards,

don, 2016-08-22:
Your Macbook has BLE, but it's not going to show up in a scan unless it's publishing a BLE service.

Try using [bleno](https://github.com/sandeepmistry/bleno) to create a BLE Peripheral. The [echo example](https://github.com/sandeepmistry/bleno/tree/master/examples/echo) is a good place to start.

tchauviere, 2016-08-22:
Hi @don

Thank you a lot for this hint !

Best regards,

closed 2016-08-22 by tchauviere