don/cordova-plugin-ble-central #302: Peripheral data with no name field

Muffasa, 2016-08-22:
when using the scan method i recive devices, but I cant see the name property in most of them.
anything i can do?

don, 2016-08-22:
I think this is a duplicate of #294. Many devices don't advertise names. 

Try the app referenced in this comment https://github.com/don/cordova-plugin-ble-central/issues/294#issuecomment-240179223

Muffasa, 2016-08-22:
thanks for your swift response!
controlling if a device is revealing his name is possible?  your other plugin BluetoothSerial setName function didnt work=/

don, 2016-08-22:
iOS uses [Device Name characteristic 0x2a00](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.characteristic.gap.device_name.xml) on the [Generic Access service 0x1800](https://developer.bluetooth.org/gatt/services/Pages/ServiceViewer.aspx?u=org.bluetooth.service.generic_access.xml) for the name. Android uses the GAP [Local Name (0x09)](https://www.bluetooth.org/en-us/specification/assigned-numbers/generic-access-profile).

Some peripherals may allow you to write to Device Name characteristic 0x2a00, but write is optional. Unfortunately iOS does not expose the GAP service.

To change the advertised name, you probably need service on the peripheral that allows this.

Muffasa, 2016-08-23:
Your other plugin BluetoothSerial do return names.
Im sorry but i cant understand why its not possible to mimic the device settings bluetooth scanning.
no need to set name, only to see the names of peripherals on ios. any possible solution?

don, 2016-09-17:
I'm not sure which platform isn't giving you names. If it's Android you might be able to read the name by connecting and reading the 0x2a00 from 0x1800. If iOS is the problem, you may be able to parse the name from the advertising info.

If you have a working sample code in github and more details about the devices and which phones are having problems, I could take a look.

closed 2016-09-17 by don