don/cordova-plugin-ble-central #305: BLE polling command

harshaddevale-tudip, 2016-08-31:
How should i send polling command to the NFC device using BLE.

harshaddevale-tudip, 2016-09-02:
Any update here please?

don, 2016-09-17:
You can poll a characteristic by continually [reading](https://github.com/don/cordova-plugin-ble-central#read) it, but it's usually better to subscribe to changes *if* the characteristic supports notification. See [ble.startNotification](https://github.com/don/cordova-plugin-ble-central#startnotification).

The commands you send the the device are device specific. You'll need to get that from the ACR1225 documentation. When you send data from JavaScript you need to send an ArrayBuffer. See the examples for [ble.write](https://github.com/don/cordova-plugin-ble-central#write) for ideas on how to format your data.

closed 2016-09-17 by don