don/cordova-plugin-ble-central #306: Notification when the phone sleeps

locsim, 2016-09-04:
Hi all,

Thanks to this great plugin, our app is able to connect to a ble device and is notified when the device button is pressed.
It works well, excepted when the phone is in standby mode...then, the button does not trigger anything anymore (the callback is fired, but after the phone has waken up, and of course this is not what we want...)

Is there any way to solve this?
Thank you,
Philippe

locsim, 2016-09-14:
Hi it's me again. I tried so keep the app listening and prevent it from falling asleep (eg by reading a value from the ble device at a given interval), but noway.
After some time, the notification is not raised anymore.
(When the phone is plugged, the notification is always raised)
This is a real issue.

Any help very welcomed!
Philippe

don, 2016-09-17:
@locsim Android notifications should work in the background. With iOS, you'll need to add [extra configuration to run in the background](https://github.com/don/cordova-plugin-ble-central#background-notifications-on-ios). I have a sample project for background scanning and notifications https://github.com/don/ble-background.

locsim, 2016-09-20:
Hi Dan, thank you for your answer.
I'm dealing with android only. I'm not sure it is a matter of background or foreground. 
My test app is waiting for a notification triggered when the ble device button is pressed. The result is the phone vibrates for 1 second.
The fact is that after some time, the vibration does not occur anymore. I mean it does not occur at all (it is not just delayed of a few seconds). When the on/off button is pressed, it works ok again for a few minutes. But only a few minutes.
It works perfectly well when the phone is plugged, even when left for hours on the desk with no action.

Any help appreciated!
Philip

closed 2017-03-16 by don