don/cordova-plugin-ble-central #309: Cordova Scan on iOS Returns Two Devices.

jcrosby10, 2016-09-12:
I'm scanning for BLE devices for an app that must run on iOS and Android. When I run the scan, it works fine but when I run it on iOS I get 2 devices. I only have 1 device that I am working with so I can't figure out why its finding the device twice.

I have a bluetooth.js file with this for scanning:
```javascript
startScan: function(services, timeout, success, failure) {
    ble.scan(services, timeout, success, failure);
},
```
At the end of my device ready function I do this:
`bluetooth.startScan([], 10, bleListeners.instrumentFound);`

And this is where I handle the scan results:
```javascript
var bleListeners = {
    instrumentFound: function(instrument) {
        if (instrument.name && instrument.name.indexOf("miniGAC") > -1) {
            console.log("Successfully found durring scan " + instrument.name + " with id " + instrument.id);
            
            var resultDisplay = document.getElementById('instruments');
            var ins = JSON.parse(localStorage.getItem("connectedInstruments"));

            var insName = instrument.name;
            for (var i = 0 in ins) {
                if (ins[i].instrumentAddress == instrument.id) {
                    insName = ins[i].instrumentName;
                }
            }
            // i added the if check to try and not add a button if there is already one there with
            // this id but the device is still added to the display twice
            if (resultDisplay.innerHTML.indexOf(instrument.id) < 0) {
                resultDisplay.innerHTML += "<button id=\"" + instrument.id + "\" value=\"instrumentButton\" style=\"background-color:#BABABA\">" + insName + "</button>";
            }
        }
    }
};
```

I also tried using `scanWithOptions` to explicitly set `reportDuplicates` to false but that did not work either. I only have this problem on iOS, it works as expected on Android. Every time I load this page on iOS two buttons are added to the HTML? Does anyone know why my one BLE device is being reported twice?

DaniloCodecoCarvalho, 2016-09-12:
When I scan for devices in IOS always return two devices either. In fact, is the exactly same device showed twice. Same name, rssi and all information.
I don't know why this happen. Android works fine.

Sometimes it returns rssi = 127 (positive) too, which is a known IOS problem.

don, 2016-09-17:
*Some* devices return 2 scan records for iOS. I'm guessing this might be related to the scan and scan response, but I'm not sure. Typically I handle this in my client code by putting the scan results into a map keyed by id.

jcrosby10, 2016-09-19:
I was able to get my code to not add the button if its already. So this isn't as big of a problem as before. So it sounds like an issue with the iOS code then, returning two responses sometimes @don ?

don, 2016-09-19:
Yes, this is something that iOS is doing. The plugin just passes the scan info from iOS to JavaScript.

jcrosby10, 2016-09-19:
Cool thanks for letting me know. This can probably be closed then I think.

closed 2016-09-19 by jcrosby10