don/cordova-plugin-ble-central #310: is ble.scan stopped when app is killed

hervethouzard, 2016-09-16:
Hello,

I'd like to know if a ble.scan() is stopped when the application is closed?
Idem, if I am connected to a peripheral, does the fact to close the app will disconnect the app?

bye,
Hervé

don, 2016-09-17:
On Android, scan should work in the background. On iOS, you'll need to add [extra configuration to run in the background](https://github.com/don/cordova-plugin-ble-central#background-notifications-on-ios). I have a sample project for background scanning and notifications https://github.com/don/ble-background

hervethouzard, 2016-09-19:
Hello Don,

Thank you very much for your answer.
In fact I wanted to know if the scan will be (well) stopped when the application is closed.
I don't want to see it running in the background

Bye

don, 2016-09-19:
iOS should stop the scan in the background unless you explicitly allow background scanning.

If you want to be sure add an event handler for [pause](http://cordova.apache.org/docs/en/latest/cordova/events/events.html#pause).

    document.addEventListener("pause", onPause, false);

    function onPause() {
        ble.stopScan();
    }

hervethouzard, 2016-09-20:
Hello Don,

thank you very much for your help.
This ticket can be closed.

Bye

apperside, 2016-09-22:
Hi, I did not understood if it works even if I kill the app, do it? Many thanks

closed 2017-01-04 by don