don/cordova-plugin-ble-central #311: iOS 10 bLE only works when application is in the background.

ds949, 2016-09-19:
iOS 10 bLE only works when application is in the background.

As such an ionic app using bLE can only perform bLE actions when changing apps, or viewing notifications/bottom settings pane with this plugin.

This behaviour has been noted on an iPhone 6s and 7, both running iOS 10.0.1

This behaviour was not observed on iOS 9.x

don, 2016-09-19:
There might be some problems with iOS 10, but I have Cordova and Ionic 2 apps that work with iOS 10 and notifications.

Can you create a simple non-ionic example project that demonstrates the problem?

lolek27, 2016-10-21:
Hi Don,

I experience the same behaviour on ios 10, and it has not been there on ios9. Bluetooth plugin only starts scanning and discovering when i go into background mode. But even when there are devices discovered it cannot connect to them. I use ionic version 1.7.15 and latest version of ble plugin.

dirtyrobinson, 2016-12-04:
This is interesting because I can't get this plugin to work in the background under iOS 10.  I followed the instructions for getting the plugin to scan in the background and used the Blefruitle example and it scans, connects and sends data just fine while the app is in the foreground but once I leave the app, it no longer functions.  May I ask to see some of your code?

don, 2017-01-04:
@ds949 @lolek27 I've been using this plugin on iOS 10, so I'm guessing that this is an application specific problem. I can't track it down without sample code, I'd prefer if you could duplicate this in a simple project without Ionic. But if you have a **simple** Ionic project that might be OK.

don, 2017-01-04:
@dirtyrobinson see https://github.com/don/ble-background

airbly, 2017-01-17:
Hi all,

I too am experiencing this issue.  I am also experiencing the same issue with the cordova-plugin-camera plugin as well.  So I am not sure it is an issue with this BLE plugin.  

To describe the issue, if I click a button in my application to take a photo or start a ble scan, the device just sits there.  If I then double tap the home button to bring up the application list and then bring the application to the foreground again, everything works for a few seconds before becoming "stuck" again.

It appears that I can repeat this indefinitely to almost "step" through the plugin's code.

This issue does not exist on iOS 9.x devices nor on any version of Android I have tested.  It is rather specific to iOS 10.

Incidentally, I also had to define my plugin as follows to make iOS 10 happy in config.xml:

```
<plugin name="cordova-plugin-ble-central" source="npm" spec="~1.1.4">
    <variable name="BLUETOOTH_USAGE_DESCRIPTION" value="stuff" />
</plugin>
```

Any insight would be helpful.  Thank you.

airbly, 2017-01-17:
A quick update.  It appears the solution to this problem is to ensure that gap://ready is in your Content-Security Policy.

`<meta http-equiv="Content-Security-Policy" content="default-src * gap://ready file:; style-src 'self' 'unsafe-inline' *; script-src 'self' 'unsafe-inline' 'unsafe-eval' *">`

lolek27, 2017-01-17:
Yes, Content Secuiryty Policy was the true cause of that issue. I discovered it several weeks ago and changed its settings with successful results.

don, 2017-01-17:
@airbly The usage description in new with iOS 10 https://github.com/don/cordova-plugin-ble-central#ios-10

@airbly & @lolek27 Thanks for the details that the CSP was causing this problem.

closed 2017-01-17 by don