don/cordova-plugin-ble-central #312: Question: is there any way to avoid cordova-custom-config?

sparklton, 2016-09-19:
Hello,
My question is about background ble operation on iOS devices [described here](https://github.com/don/cordova-plugin-ble-central#background-scanning-and-notifications-on-ios).  

Adding **cordova-custom-config** for an additional configuration on iOS will not work with Intel XDK  which we use for the builds. I wonder if there is any other way to make this work, - switching entire build system to inject one parameter for one platform sounds quite radical. Anything else we can do to achieve the same result?

don, 2016-09-20:
To get background modes to work on iOS you need to put "`UIBackgroundModes` key with the `bluetooth-central` value in its `Info.plist` file". See [The bluetooth-central Background Execution Mode](https://developer.apple.com/library/content/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html#//apple_ref/doc/uid/TP40013257-CH7-SW6).

I use `cordoba-custom-config` because it automates my workflow. Too bad this doesn't work with XDK. You could manually edit the Info.plist file, use a Cordova hook, or potentially some other Intel XDK specific solution. 

Once you figure something out, please submit a pull request to the README.md to let others know how you solved this.

sparklton, 2016-09-20:
@don - thank you for the reference, I will take it to the other side of the pond (XDK) to figure out how to achieve the same result without the hooks they don't support. Will keep this updated.

sparklton, 2016-09-20:
OK, [there is a solution](https://software.intel.com/en-us/xdk/faqs/general#android-manifest) for bypassing the usage of cordova-custom-config with Intel XDK which doesn't support Cordova hooks.

I created a dummy plugin just as described in the link, added it into XDK build and voila iPhone BLE works in the background like a champion. Here's my plugin.xml.

```
<?xml version="1.0" encoding="UTF-8"?>
<plugin 
    xmlns="http://apache.org/cordova/ns/plugins/1.0" 
    id="ble-plist-plugin" 
    version="0.0.1"
>
    <name>Dummy plugin</name>
    <description>Adds info to iOS plist file</description>
    <license>BSD-3</license>
    <engines>
        <engine name="cordova" version=">=3.0.0" />
    </engines>

    <!-- iOS applications need additional configuration to allow Bluetooth to run in the background. -->
	<platform name="ios">
		<config-file parent="UIBackgroundModes" target="*-Info.plist">
			<array>
				<string>bluetooth-central</string>
			</array>
		</config-file>
	</platform>
</plugin>
```

closed 2016-09-20 by sparklton