don/cordova-plugin-ble-central #313: Unbale to readRSSI

quadcodes, 2016-09-21:
I'm trying to use ble.readRSSI() function to receive RSSI value of the connected BLE device. 
Even though ble.isConnected() fucntion returns true, RSSI value never returns.
```
onConnect = function(device) {
 		ble.isConnected(deviceId,
    			function() {
        			console.log(True);
    			},
    			function() {
        			console.error(False);
    			}
 		);               
                ble.readRSSI(deviceId, function(rssi) {
                       console.log('read RSSI',rssi,'with device', deviceId);
                });            
};

ble.connect(deviceId, onConnect, app.onError);
```
What could be the issue here ??

markdwolfe777, 2016-09-23:
I just want to confirm that I'm seeing the same behavior, but I believe it's limited to the new iPhone 7. I have the same app running on my iPhone 6+ and a new iPhone 7, both with iOS 10.0.2. ble.connect and ble.read work fine on the 6, but throw errors on the 7. When I call ble.connect on the iPhone 7, it appears to connect, but doesn't hit either of the success or failure callbacks. If I then call ble.isConnected it returns successfully. But, if I then try to call ble.read, it hits the error callback saying saying, "Could not find service with UUID ..." I'm seeing the same thing with "NRF Connect" and "LightBlue" scanner apps. With my iPhone 6+ I can connect and view services on my BLE peripheral, but not with my iPhone 7. Did something change with Apple's CoreBluetooth library that needs updating?

hoongoon86, 2016-10-19:
I'm seeing the same error message, "Could not find service with UUID ...". 
I'm testing with iPhone5 (iOS 9) and iPhone 6, 6 plus (iOS 10). 
But this happens not very often and happens randomly.

markdwolfe777, 2016-10-19:
Just a follow up to my previous post - there are threads on the [Apple Developer Forum](https://forums.developer.apple.com/thread/63815) and the [Nordic Developer Zone](https://devzone.nordicsemi.com/question/96205/iphone7-nrf52-sdk12s132-doesnt-connect/), which pertain to the same issue and have determined it's an iPhone 7 problem. A user on the Nordic site states, "it is clearly an iPhone 7 iOS bug in 10.0.1 and 10.0.2, and is now fixed by Apple in the 10.1 beta." So, hopefully, this will solve itself with the iOS 10.1 release later this year.

don, 2016-11-18:
@hoongoon86 @markdwolfe777 @quadcodes Is this still an issue or is it fixed in iOS 10.1.x?

markdwolfe777, 2016-11-18:
iOS 10.1 fixed the issue for my setup, which was an iPhone 7 communicating with a Nordic nRF52 BLE chipset. As far as I'm concerned, it's fixed.

hoongoon86, 2016-11-18:
I don't see the message anymore in iOS 10.1.

mebrunet, 2016-12-20:
I'm also seeing readRSSI hanging on Android, but only after I've connected a second peripheral.

Nexus 5 running 6.0.1, <preference name="android-minSdkVersion" value="23" /> (also fails with 19)

Also saw this issue only a few of my colleagues newer phones but didn't note the Android versions.

mebrunet, 2016-12-21:
From `logcat` I see that once I connect a second device and enter the loop where I read the devices' RSSIs, the second "onReadRemoteRssi" is never called. From that point forward I never see "readRssi", "readRemoteRssi", or "onReadRemoteRssi" in the logs again, and most of the ble-central-plugin's functionality seems to breakdown. The bluetooth needs to be toggled on/off to get things working again... however as soon as second devices connects, it happens again.

```
[ 12-20 17:28:45.030  3084: 3178 D/BluetoothGatt ]
readRssi() - device: CC:2F:A0:32:B7:A1

[ 12-20 17:28:45.030  2887: 2904 D/BtGatt.GattService ] 
readRemoteRssi() - address=CC:2F:A0:32:B7:A1

[ 12-20 17:28:45.033  3084: 3178 D/BluetoothGatt ]
readRssi() - device: C1:7F:98:4F:05:F4

[ 12-20 17:28:45.034  2887: 3019 D/BtGatt.GattService ]
readRemoteRssi() - address=C1:7F:98:4F:05:F4

[ 12-20 17:28:45.046  2887: 2939 D/BtGatt.GattService ]
onReadRemoteRssi() - clientIf=5 address=CC:2F:A0:32:B7:A1, rssi=-62, status=0
```

tempestwf, 2018-03-23:
Did this ever get resolved? I am having the issue too.

don, 2018-06-07:
Closing old issue.

closed 2018-06-07 by don