don/cordova-plugin-ble-central #314: [Question] is there any plan to integrate ScanFilter & ScanSettings?

hervethouzard, 2016-09-26:
Hello,

I know these "options" are specific to Android but do you intend to implement them in the plugin?

Bye

don, 2016-09-27:
ScanFilter is API 21, I'm still using the older APIs. Android support is sort of a disaster because scanning and filtering are so broken on lots of Android 4.3 and 4.4 devices.

I'd like to improve this and to have an API that's still cross platform. I don't have any immediate plans to implement. If you have ideas about what the API or implementation could look like, let me know.

hervethouzard, 2016-09-27:
Hello Don,

Thank you for the answer.
I will, may be, implement it, if that's the case I will share the code.

bye

closed 2016-11-18 by don