don/cordova-plugin-ble-central #317: Error while writing on a characteristic

TomDemulierChevret, 2016-10-07:
Hi,

When I'm trying to write data to a characteristic (with the Ionic 2 wrapper around your plugin) :  
```
let key: Uint8Array = new Uint8Array(5);
key[0] = 0x80;
key[1] = 0XBE;
key[2] = 0xF5;
key[3] = 0xAC;
key[4] = 0xFF;
BLE.write(device.id, 'fffffff5-00f7-4000-b000-000000000000', 'fffffff0-00f7-4000-b000-000000000000', key.buffer)
```

I'm getting the following error :
```
10-07 14:05:40.682: E/PluginManager(31860): Uncaught exception from plugin
10-07 14:05:40.682: E/PluginManager(31860): java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.List android.bluetooth.BluetoothGattService.getCharacteristics()' on a null object reference
10-07 14:05:40.682: E/PluginManager(31860): 	at com.megster.cordova.ble.central.Peripheral.findWritableCharacteristic(Peripheral.java:559)
10-07 14:05:40.682: E/PluginManager(31860): 	at com.megster.cordova.ble.central.Peripheral.writeCharacteristic(Peripheral.java:525)
10-07 14:05:40.682: E/PluginManager(31860): 	at com.megster.cordova.ble.central.Peripheral.processCommands(Peripheral.java:637)
10-07 14:05:40.682: E/PluginManager(31860): 	at com.megster.cordova.ble.central.Peripheral.queueCommand(Peripheral.java:611)
10-07 14:05:40.682: E/PluginManager(31860): 	at com.megster.cordova.ble.central.Peripheral.queueWrite(Peripheral.java:582)
10-07 14:05:40.682: E/PluginManager(31860): 	at com.megster.cordova.ble.central.BLECentralPlugin.write(BLECentralPlugin.java:403)
10-07 14:05:40.682: E/PluginManager(31860): 	at com.megster.cordova.ble.central.BLECentralPlugin.execute(BLECentralPlugin.java:179)
10-07 14:05:40.682: E/PluginManager(31860): 	at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:117)
10-07 14:05:40.682: E/PluginManager(31860): 	at org.apache.cordova.CordovaPlugin.execute(CordovaPlugin.java:98)
10-07 14:05:40.682: E/PluginManager(31860): 	at org.apache.cordova.PluginManager.exec(PluginManager.java:133)
10-07 14:05:40.682: E/PluginManager(31860): 	at org.apache.cordova.CordovaBridge.jsExec(CordovaBridge.java:59)
10-07 14:05:40.682: E/PluginManager(31860): 	at org.apache.cordova.engine.SystemExposedJsApi.exec(SystemExposedJsApi.java:41)
10-07 14:05:40.682: E/PluginManager(31860): 	at org.chromium.base.SystemMessageHandler.nativeDoRunLoopOnce(Native Method)
10-07 14:05:40.682: E/PluginManager(31860): 	at org.chromium.base.SystemMessageHandler.handleMessage(SystemMessageHandler.java:39)
10-07 14:05:40.682: E/PluginManager(31860): 	at android.os.Handler.dispatchMessage(Handler.java:102)
10-07 14:05:40.682: E/PluginManager(31860): 	at android.os.Looper.loop(Looper.java:154)
10-07 14:05:40.682: E/PluginManager(31860): 	at android.os.HandlerThread.run(HandlerThread.java:61)
```

It seem's to be caused by `BluetoothGattService service = gatt.getService(serviceUUID);` (in Peripheral.java) not instantiating the service.
It could be caused by a bad service UUID, but `console.log(JSON.stringify(res, null, 2));` from the peripheral data returned by `connect()` shows the UUID I'm using.

Any insight that could help me solve this problem ?
Thanks in advance.

```
{
  "name": "V.ALRT A2:FE:C1",
  "id": "D0:39:72:A2:FE:C1",
  "advertising": {},
  "rssi": -32,
  "services": [
    "1800",
    "1801",
    "180a",
    "1803",
    "1802",
    "1804",
    "180f",
    "ffffffa0-00f7-4000-b000-000000000000",
    "ffffffe0-00f7-4000-b000-000000000000",
    "fffffff0-00f7-4000-b000-000000000000",
    "ffffccc0-00f7-4000-b000-000000000000"
  ],
  "characteristics": [
    {
      "service": "1800",
      "characteristic": "2a00",
      "properties": [
        "Read"
      ]
    },
    {
[... 355 lines cut ...]
      "characteristic": "ffffccc2-00f7-4000-b000-000000000000",
      "properties": [
        "Write"
      ],
      "descriptors": [
        {
          "uuid": "2901"
        }
      ]
    },
    {
      "service": "ffffccc0-00f7-4000-b000-000000000000",
      "characteristic": "ffffccc3-00f7-4000-b000-000000000000",
      "properties": [
        "Write"
      ],
      "descriptors": [
        {
          "uuid": "2901"
        }
      ]
    }
  ]
}
```

don, 2016-10-07:
It looks like you have service and characteristic UUIDs switched.

    let serviceUUID = 'fffffff0-00f7-4000-b000-000000000000';
    let characteristicUUID = 'fffffff5-00f7-4000-b000-000000000000';
    BLE.write(device.id, serviceUUID, characteristicUUID, key.buffer);

TomDemulierChevret, 2016-10-07:
Oh.
I'm definitely dumb.
I was convinced that the arguments order was characteristic then service.

It's working like a charm now.
And I'm gonna hang myself.

don, 2016-10-07:
Glad it's working.

closed 2016-10-07 by don