don/cordova-plugin-ble-central #319: ble.isConnected slow response

Gonoshift, 2016-10-16:
Hello Don, great work on this plugin!
I've ran into an annoying issue, but I don't know if the following behavior is expected or not.
If I connect to a device from Android (lets say and Arduino with a ble module), everything works as expected, but if I reset the device or turn if off, calling ble.isConnected reports a connected status for several times before finally reporting the correct disconnected status, why is that?

I get an output like:
connected
connected
connected
...
connected
disconnected
disconnected
...

Is it possible to have a faster reply to status change?

winfinit, 2016-11-14:
@Gonoshift were you able to track this down?

Gonoshift, 2016-11-14:
@winfinit So far it seems that either BLE modules (hw) have this timeout inside, that dictates after how many seconds the connection is considered non responsive/offline. The timeout could be changed in Nordic chips but they have to be reflashed with the new parameter inside. I didn't have the chance to try yet btw...

don, 2016-11-18:
I don't think there's much you can do about this. 

Rather than repeatedly calling `ble.isConnected` I'd recommend writing for the the plugin to tell you the device disconnected. The failureCallback from `ble.connect` is a long running callback and will be called whenever the peripheral unexpectedly disconnects.

Sometimes this is immediate. Other times there is a delay before Android sees the disconnect.

[ble.connect(device_id, connectSuccess, connectFailure);](https://github.com/don/cordova-plugin-ble-central#connect)

Gonoshift, 2016-11-29:
Hey @don you mean that the long running callback replies faster (even if still with a delay) than the `ble.isConnected` ?

don, 2017-01-04:
@Gonoshift some devices still have a delay when they disconnect, but I think you'll get notified faster using the callback.

closed 2017-01-04 by don