don/cordova-plugin-ble-central #320: Error in android emulator

krrishnayak, 2016-10-17:
Hello Don, Great job with this example.. however i was wondering if you cud help me out with an issue..
Getting the error when running the app in android emulator..
ERROR: Attempt to invoke virtual method 'boolean android.bluetooth.BluetoothAdapter.isDiscovering()' on a null object reference.. thnks in advance..

jensstruemper, 2016-11-05:
I think there's no Bluetooth support in the android emulator. There are people who setup android in a virtual box and attach a BLE stick to it but I haven't found an up to date tutorial for this.

don, 2016-11-05:
Unfortunately the emulator doesn't support Bluetooth. You need to run on a real device. 

@jensstruemper if you run across instructions for the virtual box setup, it would be great if you send me a link.  That's an interesting approach.

closed 2016-11-05 by don