don/cordova-plugin-ble-central #323: Call to startStateNotifications hangs in Developer app.

mebrunet, 2016-10-20:
I'm not sure what's happening here, but my calls to startStateNotifications are hanging on both iOS and Android. I have a button that triggers:

```
console.log('here')
ble.startStateNotifications(function(state) { console.log(state); });
console.log('done')
```

In the Phonegap Developer app, when I click the button I see 'here' but not 'done' in the console. Seems to work fine when I build the app out. Any thoughts?

hoongoon86, 2016-10-24:
Do you get any error messages? or do you see anything regarding state when you change bluetooth state?

mebrunet, 2016-10-24:
Nothing unfortunately. I haven't built the developer app from source (i.e. it's downloaded from the stores) so I can't remote debug it. I'll give that a try to see if I get anything more helpful.

don, 2016-11-18:
Are you seeing this on iOS or Android? What phone and OS version are you running? What version of the PhoneGap Developer App?

mebrunet, 2016-11-18:
I was seeing it on both iOS and Android... Nexus 5 Android 6.0.1, and iPhone 5 running iOS 10.0.2. Developer app is 1.7.2 now, and I'm pretty sure that's where it was at when I tried it before.

I'm just using a setInterval with ble.isEnabled to recreate the functionality. Which works fine.

don, 2016-11-18:
It's possible the PhoneGap Developer App is using an older version of the plugin. I'll try and duplicate and talk to the Adobe team if it's a version issue.

ddugue, 2017-10-30:
They are currently using 1.1.4:
[](https://github.com/phonegap/phonegap-app-developer/blob/master/config.xml#L67)

I think you can close this issue

closed 2018-06-07 by don