don/cordova-plugin-ble-central #326: ble plugin can not scan on android6.0

wzhsh90, 2016-10-26:
Hi:
 my plugin version:1.1.1  
 mobile os version: android 6.0
 
pre-condition:
 access permssionn
  <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
  <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />

issue:
  bluetoothAdapter.startLeScan return true but ble not work .I find console log BluetoothAdapter( 7682): startLeScan(): null

solution: 
   Settings -> Location, I have it turned On and then it work well.

why not we ask user open Location or auto open Location before useing ?

hoongoon86, 2016-10-26:
[https://github.com/don/cordova-plugin-ble-central/issues/242](url)
Please refer this link. 
BLE scanning in Android 6 requires location permission.
If you don't want this, I believe you should restrict sdk to 21 or lower.

don, 2016-11-05:
@wzhsh90 this is an Android restriction so there's not much the plugin can do. 

@hoongoon86 has a good idea to target the older android-21 sdk. Please update this issue if it works.

closed 2016-11-05 by don