don/cordova-plugin-ble-central #327: I can not scan my phone from the tablet using BLE

hichamrifai, 2016-11-02:
Hi,
I have been trying to run the  ble.scan() to scan for devices around. I am able to detect iphone and beacons but not android devices.  I can only detect my phone if I use a beacon simulator on the android phone.  Did I miss a small detail like the order of adding the plugin?  could you help me with that?
Thanks
Regards
Hicham

don, 2016-11-05:
The plugin will only find your tablet if the tablet is advertising a Bluetooth LE service. I think the nRF Connect app will advertise services on newer Android devices.  See #292 for more details.

closed 2016-11-05 by don

hichamrifai, 2016-11-07:
Thanks for your reply! does this applies even if the phone and the tablet are paired.  
Thanks
Hicham

don, 2017-01-05:
@hichamrifai pairing is typically for Bluetooth Classic. Bluetooth low energy devices need to advertise a service so it can be discovered.