don/cordova-plugin-ble-central #329: Repeating Bad Connection Callback on Android 6.0.1

AyeyerMa, 2016-11-16:
Hello,

When making a connection to our BLE devices on Android 6.0.1, the connection failure callback is being called repeatedly before the device has a chance to connect. This causes excessively long delays when trying to connect.

This doesn't happen on Android 5.1.1 or iOS 10. 

I also tested an earlier version of the plugin (v1.1.0) and the same problem occurs on Android 6.0.1.

Many thanks

don, 2016-11-18:
Maybe this is an Android permission problem? What error are you getting in the failure callback?

Android 6 requires location permission for BLE scanning. This means you need to grant permission to the app, have Bluetooth turned on, and have GPS turned on (on some devices.) Make sure you're using the latest version of the plugin with android-23 API support.

You might be able to target android-22 when you compile and avoid some permission problems.

AyeyerMa, 2016-11-18:
Hi Don,

Yeah, I had been aware of the permission issues you mention and we do follow them. I have reverted to Android-22 but unfortunately some of our functionality relies on 23. 

More often than not, devices will fall into the connection error callback with the following message:

{"name":"ICON_MK_R","id":"FF:3E:E2:EF:B9:BC","errorMessage":"Peripheral Disconnected"}

It seems unpredictable, sometimes a device will connect immediately and other times it will throw the above error repeatedly. Some of our customers on Android 6 have mentioned this happening also on a range of different Android phone models. 

I'm hoping it's something fixable and not a hardware issue.

Many thanks

don, 2016-11-18:
Just to clarify, does this problem go away if you target android-22?

It's odd that you'd get the error callback called multiple times. Ideally the failure callback is called 1x. Are you reconnecting between each error message?

The "Peripheral Disconnected" message comes from https://github.com/don/cordova-plugin-ble-central/blob/99d5d6b0778ac0d092373040a1303248c4f2ba32/src/android/Peripheral.java#L207-L225
Maybe there are some devices sending a bunch of intermediate states that are being interpreted as disconnected? Can you add some logging to see if you're getting bunch of state changes?

AyeyerMa, 2016-11-21:
Android-22 suffers the same issue. I've tried connecting devices in total isolation also with the same issue.

In terms of code. Very basically, what I'm doing is this when the user hits the connect button in the app:

Call `ble.isConnected()`.
If `ble.isConnected returns false`, call `ble.connect()`.
If `ble.connect() fails` call `ble.connect()` again.   

Here is some log output, it shows the immediate events up to the point of failure:

`BtGatt.GattService: onConnected() - clientIf=9, connId=0, address=XXXXXX`
`BluetoothGatt: onClientConnectionState() - status=133 clientIf=9 device=XXXXXX`
`BluetoothGatt: cancelOpen() - device: E7:62:93:65:AC:86`
`BtGatt.GattService: clientDisconnect() - address=E7:62:93:65:AC:86, connId=null`
`BluetoothGatt: close()`
`bt_btif : No such connection need to be cancelled`
`bt_btif : bta_gattc_mark_bg_conn unable to find the bg connection mask for: e7:62:93:65:ac:86`
`BluetoothGatt: unregisterApp() - mClientIf=9`
`BtGatt.GattService: unregisterClient() - clientIf=9`
`chromium: [INFO:CONSOLE(315)] "Peripheral *disconnected suddenly*{"name":"ICON_R","id":"E7:62:93:65:AC:86","errorMessage":"Peripheral Disconnected"}", `

Let me stress that sometimes, devices will connect immediately, sometimes it takes two or three reconnects and sometimes it can take up to 10 reconnects but on average it takes about 4 or 5 reconnects.

I hope this helps.

hoongoon86, 2016-11-24:
Not sure this might help but for status 133, perhaps you should wait for about 500ms then try to connect again. For me, it's working. 
And just to be clear, if none of your methods are working, just try clearing data from bluetooth app then try again.

AyeyerMa, 2016-11-24:
Thanks for the suggestion. Unfortunately, this doesn't work on Android 6 for me (the device is a Galaxy A3 2016). Interestingly, I implemented this exact fix a few months ago when I was having iOS reconnection issues and it works very well.

This issue seems to be getting progressively worse, it now causes otherwise reliable operation of our firmware updates over-the-air to malfunction because this relies on the device disconnecting and reconnecting. 

Everything works perfectly well on Android 5.1.1 mind you but there are quite a few of our customers on Android 6 who are now experiencing this connection issue.

don, 2017-01-05:
@AyeyerMa sorry about the delay. TBH, I'm not sure what's causing the issue. I expect it's some odd Android or peripheral issue or mismatch. Based on the log message, Android thinks the peripheral is suddenly disconnecting. Can you debug the peripheral and see what's happening?  If not, I'd look into the btsnoop_hci.log and compare working and broken devices.

Without the code or hardware, I can't do much more. If your still seeing the issue and want to talk about app and hardware debugging, send me an email.

closed 2017-01-05 by don

don, 2017-01-05:
@AyeyerMa Looking at #341, @mebrunet added code to handle devices that were in the process or connecting. Maybe isConnected() is a bad check. If you have a device that's slow to connect, and someone tries to connect 2x, it's possible isConnected() returns false and the 2nd connection attempt causes problems. You should be able to see multiple connection calls in btsnoop_hci.log.