don/cordova-plugin-ble-central #331: Port readRSSI to disconnected devices as well

motla, 2016-11-23:
Hello,

This is a feature request.
I've seen some apps that can scan BLE devices and display their RSSI in real-time to know which one is the nearest as you're moving.
Since the ble.scan function returns devices RSSI without a connection, I wonder why the readRSSI function is restricted to connected devices.
This would be a nice feature for example to make a real-time BLE radar.
Or is there a workaround?

Thanks everyone especially don for this great plugin I enjoy using!

hoongoon86, 2016-11-30:
I think it's only the matter of your preference. 
If you need RSSI value before connection, use returned value from scanning, 
or if you need RSSI after connected (for example after a certain time of device usage), use readRSSI. 
Please correct me if I'm wrong.

motla, 2016-12-01:
@hoongoon86 You're right but the difference is that ble.scan returns its success callback when a device is discovered, and the RSSI value from the returned device in the callback argument is static (it's the value when the callback is triggered). I would rather update the RSSI value in realtime, after a device is discovered but not connected, maybe like this : 
`setInterval(function(){ ble.readRSSI(device_id, success, failure); }, 1000);`

don, 2017-01-04:
@motla I think @hoongoon86 has the right approach here. Scan is the proper way to get RSSI for disconnected devices.

Use [ble.scanWithOptions](https://github.com/don/cordova-plugin-ble-central#startscanwithoptions) and specify `{ reportDuplicates: true }` for the options. This will provide you with constant updates as the RSSI changes.

closed 2017-01-04 by don