don/cordova-plugin-ble-central #333: Targeting android-24 sdk will produce and exception while scanning peripherals on Android M devices

mattex83, 2016-11-30:
Hello,

I know that this issue has been opened a couple of times and it is related to the permission policies of Android M, but I also read that it was closed on the 1.0.5 release of the plugin and it is still not working.

I tried to use the android-24 sdk and when the scan process starts (on a Marshmallow device) I get the following exception:

java.lang.SecurityException: Need ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION permission to get scan results

The android manifest already contains ACCESS_COARSE_LOCATION so it seems that the user must be prompted to confirm this permission anyway.

Reverting to android-22 resolves the issue so it's not a blocking issue but I think that is a problem to be faced in the future

Many thanks

don, 2017-01-05:
@mattex83 That error sounds like you have an old version of the plugin that's not prompting for Android 6 permissions.  Can you try this sample project?  https://github.com/don/ble-scan

I ran it on my Nexus 5X running Android 7.0. I'm using Cordova 6.4.0 on OS X 10.12.2. Cordova targets Android 7 (API 24) by default. 

    <uses-sdk android:minSdkVersion="16" android:targetSdkVersion="24" />

The first time you run the app, it should prompt you for permissions to access your location (which is required for Bluetooth LE).

[image]

mattex83, 2017-01-12:
Hi Don, 
I tried your project and it did not work on my environment but I think that I found what is the difference between our tests.
I ran it on a device running Android 6.0. I'm using **Cordova 5.4.1** on OS X 10.12.
To make the project work with android 24 I had to change manually the android project.settings file (default value is Android 22).
I think that the problem could be the cordova version, but it's a very strange behaviour indeed. 

At this very moment I am too close to a release date to update cordova on my environment, but on first days of february I will do it and try to run your project again to give you another feedback.
Thanks for your support.

don, 2017-01-12:
Cordova 5.4 is the problem. It doesn't have support for Android 6 permissions, so you need to target android-22. 

You need Cordova 6.x to target android-24 and have permissions work correctly.

closed 2017-01-12 by don