don/cordova-plugin-ble-central #335: Can't find connected device after app is closed

vohtaski, 2016-12-01:
Scenario (both on iOS and Android)
1. Scan for devices
2. Connect to the device `deviceId`
3. Close the app
4. Connect to the device `ble.connect(deviceId)`
-> can't connect

5. If I do a scan again, I can find it and connect.

Is it expected behavior or there is a bug somethere?

hoongoon86, 2016-12-02:
Connection requires information about the peripheral that you wish to connect.
When the app is closed(kill or quit the app), it might cause losing peripheral information. 
That's why you need another scan. 
However, if the app goes to the background and back on the foreground, I'm sure that you can reconnect to your device. 
Please don't get me wrong. I'm not saying that 'closing app -> open the app and connect without scanning' is impossible.

vohtaski, 2016-12-06:
@hoongoon86 
I am storing the info about the connected peripheral in the database as `deviceId`. Then I try to reconnect to it, but it does not work.

hoongoon86, 2016-12-06:
@vohtaski Hi, 
If you could save a peripheral successfully, I think you have to implement 
`retrievePeripheralsWithIdentifiers ` or
`retrieveConnectedPeripheralsWithServices ` 
to make a reconnection state. (of course these are for iOS)

don, 2017-01-04:
@vohtaski I think you're seeing something similar to #340. 

The easiest fix is to run a scan and ignore the results before connecting to the deviceId again.

You could try implementing the API calls that @hoongoon86 suggested in the connect function. There are similar ways to do this on Android. Adding these APIs could work but adds complexity. Re-scanning works for most cases.

closed 2017-01-04 by don