don/cordova-plugin-ble-central #337: ble.scan callbacks never called

vlafranca, 2016-12-08:
Hi,
First of all I need to preceise that I already managed to use the plugin succesfully. But for a few days now, ble.scan does not call callbacks anymore, no success, no failure, no console.log ...
I tryed to make a new ionic app totally empty with only the plugin and it does not work either ...
If you have any idea, I will be happy to try !

Thanks in advance.

Citygun, 2016-12-08:
Hello, 
I have the same issue, I manage to resolve the problem by using the ble.startScan and ble.stopScan function.

john-doherty, 2016-12-10:
I cant get either to work :(

vlafranca, 2016-12-11:
Even with startScan it's not working :(

vlafranca, 2016-12-12:
Hi,
i managed to get it work downgrading to 1.0.4 : cordova plugin add cordova-plugin-ble-central@1.0.4 and building project on cordova android 4.1.1 instead of 5+.

Hope this helps you.

carlst, 2016-12-13:
I am running cordova-plugin-ble-central 1.1.3 on Android 6.0.1 with August 5, 2016 patches
on a Nexus-7 2013 tablet.

Settings->Location is Off, `ble.startScan([], success, error)` fails to generate any BLE scan results.

Settings->Location On does generate BLE scan results.

See e.g. 
[http://stackoverflow.com/questions/33043582/bluetooth-low-energy-startscan-on-android-6-0-does-not-find-device](http://stackoverflow.com/questions/33043582/bluetooth-low-energy-startscan-on-android-6-0-does-not-find-devices)

One workaround is to change the SDK Version in  `AndroidManifest.xml` from:
    `<uses-sdk android:minSdkVersion="14" android:targetSdkVersion="23" />`
to:
`    <uses-sdk android:minSdkVersion="14" android:targetSdkVersion="22" />`

themao, 2016-12-20:
For me doesn't work neither with scan() nor with startScan(). Tried using version 1.0.4 and 1.1.3, target SDK 22 and 23. 

I have a few devices with enabled bluetooth nearby - both LE and regular ones. And no callback function is triggered.

And I'm using Cordova with Meteor, if it matters.

john-doherty, 2017-01-03:
any update on this?

don, 2017-01-04:
A sample project would really help track this down. 

If you're targeting Android-23 or greater, you'll need v1.0.5 or greater of the plugin, since that handles Android 6 permissions properly.

Android 6 requires location permission for BLE scans. So the behavior @carlst is seeing is the expected behavior. Disabling Location in settings breaks BLE scan. (These are Android's rules, not mine.)

don, 2017-01-05:
@john-doherty see if you can get this sample project to work https://github.com/don/ble-scan

themao, 2017-01-15:
I think I figured out what happens (at least in my situation). Neither of the callbacks was called, because there was no advertisement sent. I thought the BLE devices work like regular bluetooth (e.g. can be discovered by scanning). But turns out (I might be wrong though) that you need to send an advertisement from another device, then it gets detected by scanning device. There's a proper guide on how to do it https://github.com/randdusing/cordova-plugin-bluetoothle#peripheral-life-cycle

closed 2017-03-16 by don