don/cordova-plugin-ble-central #340: Scan needed before connecting?

Rlopez, 2016-12-22:
Hi there!
I have an app that locally stores name and address of BLE devices (as returned by Scan). In later runs, if I try to connect to the stored BLE device without performing a scan() I'll get a "Peripheral 20:91:48:DD:AF:24 not found." Doing a Scan() makes the issue goes away 

So, in essence:
ble.connect("20:91:48:DD:AF:24") //fails
ble.scan(...)
ble.connect("20:91:48:DD:AF:24") //works

I haven't seen any mention of that requirement in the docs, so I am not sure if it is a bug or intended, any advice welcome :)

bipintek, 2016-12-26:
I am also facing the same issue. Once scan() has been called, connect() works even if it is called many times.

don, 2017-01-04:
Yes, you need to call scan() once so the plugin can get a list of the devices.

closed 2017-01-04 by don

Rlopez, 2017-01-05:
wouldn't it be possible to store some extra data so that the scan call is not needed?

lastmjs, 2017-09-28:
@Rlopez I was thinking the same thing