don/cordova-plugin-ble-central #345: iOS10 ble.scan memleak

droghio, 2017-01-04:
I upgraded from v1.0.4 -> v1.1.3 for use with iOS10. Problem is when I try running a BLE scan half the time the scan is not called and the application's memory usage spikes until it crashes. Other half of the time it works fine.

I ran a trace using Xcode/Safari and `ble.scan` calls `cordova.exec(successWrapper, failure, 'BLE', 'scan', [services, seconds])` with the proper arguments, but on the Obj-C side the `scan:(CDVInvokedUrlCommand*)command` function is not called.

Instead the application locks up and Xcode reports mem usage that goes from ~50mb -> 300mb+ until it crashes with the message `Message from debugger: Terminated due to memory issue`

20gg, 2017-01-04:
try v1.0.5

don, 2017-01-04:
@droghio what version of Cordova are you using?

droghio, 2017-01-04:
@don 6.4.0

@20gg Tried v1.0.4 - 1.1.3, all have the same problem on my system. v1.0.4 worked prior to iOS10 though.

don, 2017-01-04:
@droghio please see if you can duplicate using https://github.com/don/ble-scan

droghio, 2017-01-05:
@don Thanks for throwing the template project together. While I was unable to replicate the exact problem I was able to produce odd behavior that ultimately helped me pin down the problem (or at least what seems to be the problem).

I was wrapping the `ble.scan` call inside a `ble.isEnabled` check, and this seems to be causing the crash/odd behavior. For your test script when I ran your default code my phone picked up my laptop (quartz in the log) and reported it. Typically it would only report it one time, but half the time it would pickup the same advertisement in a flood. 

When I wrapped the scan inside a `ble.isEnabled` the rate to flood vs. singleton reports skyrocketed, 10/10 trials vs 5-6/10.  I've attached logs of both cases.

When I went back and removed the `ble.isEnabled` wrapper my app seemed to work fine. Roughly 30 open/closes later and I still haven't encountered an issue (vs once every 5-6 times max before the change).

Not much to go on, and not super sure what was going on but my app seems to be working. Any idea why/if nesting the scan inside the isEnabled does/might cause issues?

Below is how I nested the two in your sample project:
```
        ble.isEnabled(function (){
            ble.scan([], 10, app.onDeviceDiscovered);
            setTimeout(app.scanComplete, 10500);
        }, function (){ div.appendChild.createTextNode('Is Enabled Failed') })
```

Log Files:
[logMultiple.txt](https://github.com/don/cordova-plugin-ble-central/files/686554/logMultiple.txt)
[logSingle.txt](https://github.com/don/cordova-plugin-ble-central/files/686555/logSingle.txt)

don, 2017-03-07:
@droghio not sure why that causes a problem

Consider using [ble.startStateNotifications](https://github.com/don/cordova-plugin-ble-central#startstatenotifications) to track the Bluetooth state.

closed 2017-03-16 by don