don/cordova-plugin-ble-central #347: Mobile app doesn't detect every iBeacon advertisment

MarinRadmilovic, 2017-01-13:
Hey! This plugin is great and very easy to use. Congrats on that. I made a simple application that detects iBeacon when near and everything works fine except one thing. I use startScanWithOptions and reportDuplicates is set to true. I set beacon advertising frequency on 500ms. The problem is that startScanWithOptions callback is called every 5-6 seconds for one beacon, and not every 500ms. Is it problem with Bluetooth hardware or with this plugin, and is there any way to get every message. Thank you very much and keep up the good work.

delkon, 2017-02-12:
@MarinRadmilovic Please could you share your approach for getting iBeacon information!
I need to retrieve it too but instead of UID, Major, Minor have received only standard Adv. data.

Thanks for any info

don, 2017-03-16:
Android should give you a notification when the beacon is scanned. Try another tool/device to determine if the beacon is not scanning or it's an Android issue. 

Note that you're not going to get any beacon info on iOS with this plugin. Apple strips the beacon advertising information and you need to use CoreLocation instead of CoreBluetooth.

You might want to take a look at a beacon specific plugin https://github.com/petermetz/cordova-plugin-ibeacon

closed 2017-03-16 by don