don/cordova-plugin-ble-central #350: No kCBAdvDataManufacturerData.data on iOS

lcrocker, 2017-01-24:
My project needs to get the Manufacturer's advertising data from a device. This was previously working by getting kCBAdvDataManufacturerData.data from the returned peripheral and parsing the bytes manually, which I had no problem with. But now this returns undefined. Is the Manufacturing data hiding under another name now perhaps? JSON.stringify(kCBAdvDataManufacturerData) doesn't show any useful info either.

lcrocker, 2017-01-24:
I had previously seen this work with version 1.1.2 of the plugin, but now it no longer works there either, so I suspect the change is on the Apple side (upgrade of iOS or Xcode or something). I'll investigate more. FYI, I'm currently using Xcode 8.2.1, and installing on an iPad mini with iOS 10.2.

lcrocker, 2017-01-24:
Seems to be the function at line 390 in BLECentralPlugin.m. There are NSLog statements there that look useful. I tried building the app with "cordova build --debug", but that doesn't seem to activate them (at least nothing shows up in /private/var/log). Is there a way to activate those NSLog traces so I can help debug this?

lcrocker, 2017-01-24:
Got it into Xcode and compiling for Debug shows me the NSLog()s and my own console.log()s. As you can see below, the manufacturer data does come in (it appears to be base64 encoded: "EAQIEcQW" for 6 bytes). The last line is the result of console.log(p.advertising), and the data is gone.  I'm sure it's in the function I mentioned above, but I'm not much of an ObjC guy, so I'm not sure I'll be able to fix it.

2017-01-24 14:48:45.008662 BluTest[859:278412] Discovered {
    advertising =     {
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataLocalName = BAPI7002;
        kCBAdvDataManufacturerData =         {
            CDVType = ArrayBuffer;
            data = EAQIEcQW;
        };
        kCBAdvDataServiceUUIDs =         (
            "6E400001-B5A3-F393-E0A9-E50E24DCCA9E"
        );
    };
    id = "2A7F5D5C-F704-49F6-912D-1640E8AA0C70";
    name = BAPI7002;
    rssi = "-69";
}
2017-01-24 14:48:45.030171 BluTest[859:278412] {"kCBAdvDataLocalName":"BAPI7002","kCBAdvDataManufacturerData":{},"kCBAdvDataServiceUUIDs":["6E400001-B5A3-F393-E0A9-E50E24DCCA9E"],"kCBAdvDataIsConnectable":true}

lcrocker, 2017-01-25:
Ah, the light has finally dawned!  Apparently, now Cordova uses the "plist" library to make Apple's NSDictionary into a javascript object with fancy types, including that "ArrayBuffer" above. So the "kCBAdvDataManufacturerData" object is now itself an ArrayBuffer; it no longer contains a "data" field to be base64-decoded; that's already been done, and one just has to wrap the ArrayBuffer in something like a Uint8Array() to get the bytes out of it.

So, let's call this an exercise in minimal documentation all around, but especially cordova-platform-ios. Might want to add something about this to your docs as well.

don, 2017-02-02:
@lcrocker thanks for tracking this down. I'm not sure if Cordova or iOS changed here but I agree it needs better documentation.

Sureshee, 2017-06-30:
@lcrocker @don, I have two question on parsing this Manufacturer data from the Scan,
I tried to parse the data field from the kCBAdvDataManufacturerData, in some device, i still see that the values are base 64 encoded.
I don't get the value if i either use Uint8Array as @lcrocker suggested, or when i tried decoding the base 64 value using a separate function as shown below, can you help me in showing a template on how to do it.
private base64ToBase16(base64) {
  return window.atob(base64)
      .split('')
      .map(function (char) {
        return ('0' + char.charCodeAt(0).toString(16)).slice(-2);
      })
     .join('')
     .toUpperCase();
} 

	console.log('data:', this.base64ToBase16(device.advertising.kCBAdvDataManufacturerData.data)); 

Also What does CDVType represents, I see only in some devices, which is a arraybuffer.

2017-06-30 11:22:40.182444+0200 MyApp[598:112266] bleScanner.startScan
2017-06-30 11:22:40.182655+0200 MyApp[598:112266] startScan
2017-06-30 11:22:40.236627+0200 MyApp[598:112266] Discovered {
    advertising =     {
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataLocalName = "CC2650 SensorTag";
        kCBAdvDataManufacturerData =         {
            CDVType = ArrayBuffer;
            data = "DQADAAA=";
        };
        kCBAdvDataServiceUUIDs =         (
            AA80
        );
        kCBAdvDataTxPowerLevel = 0;
    };
    id = "F674D953-925E-4371-B273-3468D481E936";
    name = "CC2650 SensorTag";
    rssi = "-43";
}
Looking forward to hear your suggestions

lcrocker, 2017-06-30:
Don't use the ".data" property:

    mdata = new Uint8Array(d.advertising.kCBAdvDataManufacturerData);

valterforesto, 2017-10-26:
@don @lcrocker,
I'm currently using the "cordova-plugin-ble-central" spec="^1.1.4" on iOS 10 and on iOS 11.
The 'kCBAdvDataManufacturerData' it's not any more reported on the 'd.advertising' object returned by scanning an iBeacon 

```
[phonegap] [console.log] device: {"id":"590C9F62-180B-4799-93F4-04700F6B55CF","rssi":-39,"advertising":{"kCBAdvDataIsConnectable":
false}}
```

In the past I used the "cordova-plugin-ble-central" with iOS for scanning iBeacons and the 'kCBAdvDataManufacturerData' was correcly reported on 'd.advertising' object returned when scanning an iBeacon. Can you see the same behaviors on yours devices / apps ?

Thanks in advance for yours replies. 
Valter.

don, 2018-06-07:
@valterforesto iOS hides beacon information from CoreBluetooth. You really need to use CoreLocation with beacons on iOS. Try https://github.com/petermetz/cordova-plugin-ibeacon

don, 2018-06-07:
Fixed by 7c3d4ae06d4087620eb80fc80053cbfa6e85d929 (Issue number in the commit message is incorrect.)

closed 2018-06-07 by don