don/cordova-plugin-ble-central #351: Reg BLE.startScan()

santosh532, 2017-01-25:
Great work with the plugin !!

Using plugin within an ionic 2 application. 
Ionic info 
`
Cordova CLI: 6.4.0 
Ionic Framework Version: 2.0.0-rc.3
Ionic CLI Version: 2.1.12
Ionic App Lib Version: 2.1.7
Ionic App Scripts Version: 0.0.46
ios-deploy version: 1.7.0 
ios-sim version: 4.1.1 
OS: macOS Sierra
Node Version: v4.1.2
Xcode version: Xcode 7.3.1 Build version 7D1014`

I find that the **ble.startScan** identifies any new beacon that comes into the proximity but does not update the RSSI value does not update the RSSI value realtime on when the mobile device is on the move. Is there any way that plugin handles this / should there be a timeout, which keeps running in the background and scans for the beacon's updated RSSI value.

don, 2017-02-02:
You need to use [ble.startScanWithOptions](https://github.com/don/cordova-plugin-ble-central#startscanwithoptions) and pass [{ reportDuplicates: true }](https://github.com/don/cordova-plugin-ble-central#quick-example-2) for the options, to get updated data.

If you are doing lots of beacon stuff you might want to check out https://github.com/petermetz/cordova-plugin-ibeacon

closed 2017-03-16 by don

tearforfear007, 2017-11-09:
I have `ble.startScanWithOptions` and `{ reportDuplicates: true }` for the options, but how can I refresh my List to get the updated rssi for each ble?? It scan a lot of device, but I need just need to update my rssi ………T^T

ddugue, 2017-11-09:
If you only need the RSSI of one device, you could always connect to it and use `readRSSI`

The way with `startScanWithOptions` is to create an object map and update Rssi on it with the scan records and simply ignore the other data.

Hope it helps,