don/cordova-plugin-ble-central #352: How to keep the connection even after app is closed or phone rebooted?

R2D221, 2017-01-27:
Hi, I'm currently making an app that requires to be connected to a Bluetooth device (essentially just a button) at all times, so that whenever I press the button, I can run a callback. I need this connected at all times, so that unless it's explicitly disconnected by the user, the connection is active (and try to reconnect if the connection is lost, e.g., because of distance). It should also handle a reboot, but if it's not possible, then I'm OK with opening the app again.

Although I have experience with Cordova, so far I haven't made any background service, so I'm wondering if this can be done within Cordova, and if your plugin supports such usage.

I'm creating an app for both Android and iOS.

Thanks in advance.

don, 2017-02-02:
Android can scan and receive notification in the background.

For iOS you'll need some additional configuration see [Background Scanning and Notifications on iOS](https://github.com/don/cordova-plugin-ble-central#background-notifications-on-ios).

Your app will need to be responsible for scanning again when a connection is lost or scanning when the app starts.

R2D221, 2017-02-02:
“Android can scan and receive notification in the background”

... Just like that? Isn't there anything I need to do in order to avoid my
app getting killed by the system?

don, 2017-02-02:
I think so. Try this sample app to test the behavior https://github.com/don/ble-background

closed 2017-03-16 by don

markarupert, 2018-11-30:
Hi Don,

> Android can scan and receive notification in the background.
>
> For iOS you'll need some additional configuration see [Background Scanning and Notifications on [...]

I looked at the background example but I don't fully understand. Is the following referring to a virtual device that I can use or does the device that I am working with have to have a counter service or I am out of luck

"This projects includes and Arduino version of the counter service that runs on the Arduino 101 or any board supported by blePeripheral. An experimental Node.js verion of the service is also available."