don/cordova-plugin-ble-central #353: Add link to fork to README

tomasbedrich, 2017-02-04:
Hello @don, I have created a fork extending your code with Nordic DFU capability. There is a new method `upgradeFirmware` which allows users to upgrade nRF5x based chips over the air.

https://github.com/fxe-gear/cordova-plugin-ble-central

Others may be interested in this capability too, but I don't want to create a PR, because the functionality is a little bit out of scope of your original code. The alternative may be placing a link somewhere on the bottom of the README, what do you think?

If you think it is not interesting for you nor the others, please close the issue. Thank you for response

don, 2017-02-04:
@tomasbedrich this looks great. I'm thinking this could be packaged as an OTA plugin that depends on my plugin. I'll get stuff working on my hardware and let you know.

closed 2017-08-22 by don

logindharam, 2018-04-19:
how to upgrade the upgradeFirmware of nordic device I am not able to connect with ble plugin

commit 8088d973 (2017-08-22): Nordic DFU

Fixes #353