don/cordova-plugin-ble-central #354: startNotification not receiving notifications from BLE device

grubicv, 2017-02-04:
I have a device that should send updates to a device registered to receive certain characteristics. I've wrote a Swift native code for it as a standalone app, and it works, but when I try this plugin using Phonegap, it's not working. I've used newest cli, and cli-6.1.0, but the issue still remains the same.
I can scan for devices with specific service UUID, I can connect to them, read characteristics, but if I register to be notified using startNotification I am not receiving notifications for some reason (and also there is no error when registering for it).
I also tried to compile the app on the Phonegap website, instead of using phonegap developer app, and same issue persists. I've tested it on an iOS device with Bluetooth 4.2, and Android device with 4.2 - both work flawlessly with the tracker with their native apps implementing same principles I've tried using this plugin.

Any ideas what am I doing wrong? I suspect something regarding the build does not function properly, or something specific with startNotification is not working (have you retested the example apps after changing to startNotification from old functions?)...

grubicv, 2017-02-04:
Definitely has something to do with specifying plugin version. If I use old notify, it works, but startNotification doesn't... Is there anything needed to be specified somewhere except in config.xml?

don, 2017-02-08:
@grubicv I'm glad it's working but I'm not sure why. Version 1.1.5 doesn't exist. 1.1.4 is the latest version in NPM.

ble.notify() and ble.startNotification() call the same code.
https://github.com/don/cordova-plugin-ble-central/blob/master/www/ble.js#L124-L133

The `<gap:plugin>` syntax is for [PhoneGap build](https://build.phonegap.com). Some people hvae reported they need to also specify `<preference name="phonegap-version" value="cli-6.1.0" />` in config.xml.

Since you have Xcode and the Android SDK installed have you tried building locally with the command line? Install the plugin with 

    phonegap plugin add cordova-plugin-ble-central

Then you can debug in Xcode. If you see an error message let me know.

If there's a problem with a specific example app, let me know. I haven't run them in a while but they all should work.

grubicv, 2017-02-08:
I've installed the plugin using:
`phonegap plugin add cordova-plugin-ble-central`
and added everything I needed to config.xml.

As stated in first post - I've tried cli-6.1.0 - it didn't help. As for testing it - I will see what I can do, but there is no problem with the plugin code for iOS or Android (as you stated both notify and startNotification call the same function/method), but the issue here is that it looks like somehow the old plugin gets built into the app.

Sorry for not clarifying enough - when I stated I updated to spec 1.1.5, I've also changed it from 1.1.4 to 1.1.5 in plugin itself (in package.json and plugin.xml), and that (for some reason) did the trick.

closed 2017-03-07 by don

subratpalhar92, 2020-09-07:
make sure you are using write() instead of writeWithoutResponse(), this solved my issue on ios