don/cordova-plugin-ble-central #356: iOS didUpdateValueForCharacteristic gets triggered for only the first time

Huiting-Yeong, 2017-02-08:
I am sending a message to my ble device. I have used the android portion and it works fine. When using the plugin for iOS, after sending a message, `didUpdateValueForCharacteristic` gets triggered and I can get the reply for the first time. But subsequent sending of the same message does not trigger `didUpdateValueForCharacteristic` and I am unable to get the reply back.

Any ideas?

Thanks!

don, 2017-02-08:
[didUpdateValueForCharacteristic ](https://developer.apple.com/reference/corebluetooth/cbperipheraldelegate/1518708-peripheral?language=objc) is only going to get called when the peripheral sends a *read* response or *notification*.

Are you writing the same value to the characteristic both times?

I'd look to see what the peripheral is sending to your app. If you can't debug the peripheral code, try these alternatives. For Android, enable btsnoop_hci.log in the debug settings, then feed that into wireshark. For iOS, I use a bluetooth sniffer and wireshark.

Huiting-Yeong, 2017-02-08:
Hi don, thanks for the comment. 
What type of bluetooth sniffer should I use for ios? Did some research and found this: https://developer.apple.com/bug-reporting/profiles-and-logs/ 
Is this what I should be aiming for? 

I do not wish to purchase any sniffer or jailbreak my phone.

don, 2017-02-09:
@Huiting-Yeong I use this [BLE sniffer from Adafruit](https://www.adafruit.com/products/2269)

Based on the apple docs, you might be able to get a Bluetooth log http://adcdownload.apple.com/iOS/iOS_Logs/Bluetooth_Logging_Instructions__iPhone_and_Apple_Watch.pdf (login required). Please let me know if that works.

Huiting-Yeong, 2017-02-09:
The apple logging gives me a .logarchive file which lists down the connection details but doesn't aid me in viewing each bluetooth packet for writing messages.

I have with me a TI CC2541 chipset, will try using that as a BLE sniffer.

When running native swift (separate ble code), I start notification and my peripheral triggers `didUpdateValueForCharacteristic` every time i perform a write. Only when running through the plugin, I am getting the problems. So, suspecting the problem shouldn't be with the peripheral device.

don, 2017-02-09:
It's tough to tell without seeing code or knowing about the peripheral. This plugin just wraps the native APIs should it should behave the same. Make sure you don't use a promise for startNotification since the success callback will be called many times. Check the properties on the characteristic you're writing. Maybe you need a writeWithoutResponse instead of write?

If you have some details about the peripheral and can share the code I can take a look.

Huiting-Yeong, 2017-02-09:
Hi, was reading the documentation on apple and it says for `didUpdateValueForCharacteristic`: `This method is invoked when your app calls the readValueForCharacteristic: method, or when the peripheral notifies your app that the value of the characteristic for which notifications and indications are enabled (via a successful call to setNotifyValue:forCharacteristic:) has changed.`

Does it mean if my peripheral device were to reply with the same value,`didUpdateValueForCharacteristic` will not get triggered as the value of characteristic did not change?

don, 2017-02-22:
@Huiting-Yeong I think the problem is with your peripheral or your app. I created a test project https://github.com/don/ble-Issue356. The behavior on iOS and Android is the same for me. The test project contains a peripheral you can run on your Mac using Nodejs 6.9.5.

Tested with 
* Cordova 6.5.0
* cordova-plugin-ble-central 1.1.4
* Xcode 8.2.1
* iOS 10.2.1 on an iPhone 5s
* Android 7.1.1 on a Nexus 5x

Huiting-Yeong, 2017-02-23:
Hi don, just to update. I found out that if the code itself calls multiple commands it works. (Below example)

```
ble.writeYesCommand(..., function(result){ 
   console.log(result); 
   nextFunction();
}, function(error){});

nextFunction(){
   ble.writeNoCommand(..., function(result){
   ...}, function(error){});
}
```

But if i were to call the two commands separately through tapping two different buttons, only the first one will work. I am able to work around with the first way for now so I will be closing this thread.

Once again, thanks for all your help!

closed 2017-02-23 by Huiting-Yeong

Huiting-Yeong, 2017-11-13:
Previous did a temporary solution but I have managed to solve this issue recently.

Context:
I will be passing a message of 32bytes through BLE communication.

I would only construct the message in C (at the plugin side) so I would call as such:

`ble.write(device_id, service_uuid, characteristic_uuid, "", success, failure);
`
I modified the plugin src files for write function as such where message1 and message2 is the 32bytes messages split into 2 parts. 

           ```
 [peripheral writeValue:message1 forCharacteristic:characteristic type:CBCharacteristicWriteWithResponse];
            [peripheral writeValue:message2 forCharacteristic:characteristic type:CBCharacteristicWriteWithResponse];
```

This worked for me but only for the first command send. Subsequent commands failed to get notifications.

Solution

            ```
[peripheral writeValue:message1 forCharacteristic:characteristic type:CBCharacteristicWriteWithoutResponse];
            [NSThread sleepForTimeInterval:.00005];
            [peripheral writeValue:message2 forCharacteristic:characteristic type:CBCharacteristicWriteWithoutResponse];
```

I experimented and this way finally worked for me. I changed the write type to WriteWithoutResponse and included delay after the first write.