don/cordova-plugin-ble-central #357: No response when writing to HM-10 BLE

becauseinterwebs, 2017-02-09:
Using the [bluefruitle](https://github.com/don/cordova-plugin-ble-central/tree/master/examples/bluefruitle) example I changed the service UUID's for the HM-10 BLE module according to a BT sniffer and[ this article](http://blog.blecentral.com/2015/05/05/hm-10-peripheral/).  The HM-10 exposes two service ID's:

```
// this is HM-10's UART service
var bluefruit = {
    serviceUUID: '0000ffe0-0000-1000-8000-00805f9b34fb',
    txCharacteristic: '0000ffe1-0000-1000-8000-00805f9b34fb', // transmit is from the phone's perspective
    rxCharacteristic: '0000ffe1-0000-1000-8000-00805f9b34fb'  // receive is from the phone's perspective
};
```

The example app works great on Android, but on iOS I get the following error when trying to send data:

`Failed writing data to the bluefruit le: Could not find service with UUID 0000ffe0-0000-1000-8000-00805f9b34fb on the peripheral with UUID 800A7023-C1B9-12AC-6DAF-3384C7A8F0A2`

Since it's already connected to the device, why can't it find the main service ID?

Also, by changing the error function from:

```
var failure = function() {
     alert("Failed writing data to the bluefruit le");
};
```
to
```
var failure = function(error) {
     alert("Failed writing data to the bluefruit le :" + error);
};
```
in the sendData method you are able to see the error reason.

Thank you!

don, 2017-02-09:
Try the short 16-bit versions of the UUIDs: ffe0 and ffe1

becauseinterwebs, 2017-02-09:
I tried the following:

```
ble.write(
                    app.deviceId,
                    0xffe0,
                    bluefruit.txCharacteristic,
                    data, success, failure
);
```

and 

```
ble.write(
                    app.deviceId,
                    0xffe0,
                    0xffe1,
                    data, success, failure
);
```

And now I get no response, success or error...

becauseinterwebs, 2017-02-09:
Ok, maybe getting closer... when I use:

```
ble.write(
                    app.deviceId,
                    'ffe0',
                    bluefruit.txCharacteristic,
                    data, success, failure
);
```

I get an error "cannot find characteristic with UUID <txCharacteristic here....>

When I use: 

```
ble.write(
                    app.deviceId,
                    'ffe0',
                    'ffe1',
                    data, success, failure
);
```

I get "Writing is not permitted"

becauseinterwebs, 2017-02-09:
Ok, I have it working on iOS now.  I changed the setup to:

```
var bluefruit = {
    serviceUUID: 'ffe0',
    txCharacteristic: 'ffe1', // transmit is from the phone's perspective
    rxCharacteristic: 'ffe1'  // receive is from the phone's perspective
};
```

And it is now using "WriteWithoutResponse":

```
ble.writeWithoutResponse(
                    deviceId,
                    bluefruit.serviceUUID,
                    bluefruit.txCharacteristic,
                    data, success, failure
);
```

And it is working.  Going to see if the Android device will work as well with the setup.  if not, I'll just use a different setup for each device.

becauseinterwebs, 2017-02-09:
Update:  Yes that same settings works great with Android!  Woot!  Thank you for your help and for this great plugin!

closed 2017-02-09 by becauseinterwebs